DSheirer/sdrtrunk #937: Occasional Missed Audio

DSheirer, 2020-09-21:
Note: this was posted in user support forums: https://groups.google.com/g/sdrtrunk/c/2GmvyoSR9No
Screenshots indicate channel is decoding correctly.
---------------------
At random times the audio will not be present on SDRTrunk.

When it is working the audio quality and digital decode is the best I have experienced compared to other digital decode software.  No "digitalizing"- always good.

The first screen shot below is indicative.  I know there was audio as I was monitoring it on a desktop scanner radio. The absence of a Talk Group “Name/Number” in the sound bar may be observed. 

The second screen shot has audio working.  It indicates "Gillespie" - the agency in the sound bar.

I realize this is an old hardware configuration and a new v. of Ubuntu and a new alpha version of SDRTrunk. 

Estimate it happens about 15% of the time.

Most times it comes through. This not only happens on short transmissions but also on longer transmissions.

Running Ubuntu 20.04.1 with Gnome 3.3.3 Windowing X11

Dell Optiplex 755

8 gigabytes memory

intel Core 2 Duo CPU E8400 / 3 gHz

250 Gb hard drive but only using half as the other half is for another O/S.

Using the process of elimination, I have utilized two other new computers running windows 10 and used both the SDRTrunk V0 .4 final and the V.0.5.0-alpha3 on both computers and still about 15% of the time the audio will not be present.

SDRTrunk V.0.5.0-alpha3

RTL2832U SDR/R820T

The RTL is warmed up....after several hours.

Monitoring a P25 Phase II trunking system, non - simulcast.   770 mHz

The frequencies range from 773.281 mHz to 774.78125 mHz.

The tower is 1 miles from my location.

Using a rubber ducky with SMA connector to the RTL-SDR

(But have also used outside antenna with 50’ Times Microwave LMS 400 cable)

(Proven system to monitor 700 mHz on a desktop radio.)

The FFT display waterfall shows traffic

The spectral display shows the software has landed on the signal.

The Status shows receiving a call

The Events show Group call, duration, time etc.

I have:

Rebooted,

Used various gain settings (Incidentally one desktop radio I had would not decode if signal overload)

Tested configuration settings - using Default Mono and Default Stereo. 

Both Poly and Heterodyne on the Channelizer Type.

Max Channels from 1 to 5. 

Tried another site on this system about 14 miles away.

DSheirer, 2020-09-21:
Looking at the P25P1 audio module, there's a possibility that the LDU2 encryption sync parameters may not have passed the CRC check and may be presenting invalid info and the call audio is erroneously being flagged as encrypted by the audio module, preventing the audio from playback.

Levy8Me, 2023-10-25:
I'm also running into this issue

DSheirer, 2023-11-09:
> I'm also running into this issue

@Levy8Me I'm closing this issue because the original thread author indicated the issue was resolved by configuration/environmental changes.

Have you posted in the support forum the issue you're having to see if anyone can help you to resolve your missing audio?  If your issue cannot be resolved and we can define/identify that there is a code issue/bug, then we can open a new issue here on GitHub.

closed 2023-11-09 by DSheirer