DSheirer/sdrtrunk #938: Duplicator system seems to break randomly a few times a day.

Ctpolicelive, 2020-09-21:
I use 3 RTL-SDR's to pickup two phase2 simulcast towers. So my sdrtrunk handles a lot of local duplicates to achieve this. Using the Version 0.5.0 Alpha 3.  I feed this stream both into broadcastify as a live stream, and also to the calls system. https://www.broadcastify.com/listen/feed/28978

Running it on a dedicated linux mint laptop. Everything works great. But one issue I have been having for weeks is the duplicator seems to be bugging out for whatever reason and I have to reset the software to fix it. 

It breaks and will just play constant doubles on my live stream. Seems to happen at heavy radio traffic times. Or when Troop F and Norwich police talk. I can't predict it.. It is just so random. It happened tonight 3 times in a row. But most likely won't happen again until tomorrow afternoon.

GTR8000, 2020-09-22:
https://github.com/DSheirer/sdrtrunk/issues/905

GTR8000, 2020-09-22:
You may want to build the latest master branch, as there is a fix for this issue.

https://github.com/DSheirer/sdrtrunk/commits/master

Ctpolicelive, 2020-09-22:
Thanks a ton!! Yep that is my issue. 

I just gave it the best I could and tried to build the master on Linux Mint, Ubuntu, and even Windows 10.. All failed. Got the same java error.  (Execution failed for task ':compliejava' Could not target platform: 'Java SE 13' using tool chain: 'JDK 11 (11)'. 

Looking like I will just have to wait until the newer release of sdrtrunk comes out. I just cannot figure out how to compile the master. 

Thanks again.

liquidmayo, 2020-09-22:
I am having the same issue as described.  Happy to see a fix coming.

closed 2020-09-23 by Ctpolicelive