DSheirer/sdrtrunk #939: Change Streaming Audio Sample Rate to 22,050 kHz

DSheirer, 2020-09-22:
Broadcastify streaming archive service is removing silence and reencoding audio prior to archiving, causing poor audio quality. 

Update MP3 encoder to use 22,050 Hz sample rate.

GTR8000, 2020-09-23:
For reference:

https://broadcastify.zendesk.com/hc/en-us/articles/204740095-Setting-up-your-Broadcasting-Station

> Other packages such as SimpleCast, ProScan, Windows Media Encoder, and others should use the following settings for best results.

For one scanner:

    Channels = 1 for Mono
    Bit rate = 16kbps
    Sample rate = 22050kHz
    Format = MP3

GTR8000, 2020-09-23:
BCFY Calls should also be encoded at 22050 Hz

https://wiki.radioreference.com/index.php/Broadcastify-Calls-API#Convert_to_MP3_MPEG_Audio

DSheirer, 2020-09-29:
From google groups:
--------------------------
I would LOVE TO SEE USER OPTIONS to the encoding PER FEED. for bit rate, and sample rate.

I operate my own feeds end to end. from the source to the server used by clients(users), I have in the double digits for feeds... and when I used Darkice for the feeds it was set at

Sample Rate; 22.05KHz
Sample Bits : 16
Stream Bit Rate: 24kbs
Stream Mode: CONSTANT bit rate

The 2 remaining analog sources that come from either a radio or SDR spitting out analog, well not really its still digital bits,  one uses the FANTASTIC WolfSon audio card that stupid idiots killed bloody mergers... any way....it only likes 44.10Khz sampling.. and the audio fed there is much "richer" in my opinion.

the other feeds is NOAA NWR in my area... and so it has the above config its source is a SDR. The Pi/Woflson is still tied to a radio, for now...

In the future I'd love to see the Streaming fields have

SAmple Rate - dropdown with 44.100 and 22.05 KHz
Sample Bits: 16, 32
Bit Rates: 16,24,32,64 ( I don't know that more or less choices really make sense... ie: a 192Kbps mono ??? Really? Especially since you have to remember silence is sent at this rate too..)

Mode: Constant or Variable, ( personally I think variable is pointless, but that I think is all in the "listener")

jmp367, 2020-10-18:
I too would like to see the streaming audio sample rate set to 22,050 kHz.

closed 2022-03-10 by DSheirer

DSheirer, 2022-03-10:
@GTR8000 Added MP3 preferences to allow user to select MP3 encoder settings and audio sample rate.  This is a global setting for now and I might try to rearchitect streaming framework to allow a per-stream customization.

I listened to my Broadcastify Calls stream using the CBR 16 with 22050 sample rate and the audio sounds raspy, probably because of the resampling from 8000 to 22050 prior to MP3 encoding.  Once this release is stable, if you have a chance, could you check the Broadcastify archive audio to see how the changes affect the quality of the reprocessed audio (ie silence removal).

PR #1159 (2022-03-10): #939 Adds MP3 encoder preferences to the User Preferences dialog and …
#939 Adds MP3 encoder preferences to the User Preferences dialog and provides user control over sample rate and MP3 encoder settings.  MP3 preferences are global and affect both recording and streaming.

Resolves #939