DSheirer/sdrtrunk #943: HackRF amplifier

ah1102, 2020-09-28:
In what position is the "amplifier" button active?

In such:
[image]

or this:
[image]

In the latter case, when the button is active, the signal levels, on the contrary, fall.

TheKing505, 2020-10-09:
`src/main/resources/52-airspy.rules;src/main/resources/__````javascript
function fancyAlert(arg) {
  if(arg) {
    $.facebox({div:'#foo'})
  }:~/unrealircd$ ./unrealircd rehash
Rehashing Unreal Demon505@outlook.com~/unrealircd$ tail logs/ircd.log
[Sun Jun 28 15:03:00 2015] - TIME SYNCH: IRCd clock succesfully synchronized to known good time source.
[Sun Jun 28 15:27:42 2015] - Loading IRCd configuration ..
[Sun Jun 28 15:27:42 2015] - Configuration loaded without any problems ..
}
```15:31:28] -irc.test.net- *** Notice -- OperX is rehashing server config file
[15:31:28] -irc.test.net- *** Notice -- Loading IRCd configuration ..
[15:31:28] -irc.test.net- *** Notice -- Configuration loaded without any problems ..

ah1102, 2020-10-10:
There is an assumption that the amplifier burned out in my device and the program is not to blame.

closed 2020-10-10 by ah1102