DSheirer/sdrtrunk #946: Broadcastify Call Streaming Nor Working

andrewp-sys, 2020-10-11:
Apologies if this "bug" is a result of my error, but it appears that your software does not work with Broadcastify Call.
I entered in my API Key and System ID but the software refuses to connect, and simply returns "error". When testing the connection, it says the connection failed and the error is "null", which is not helpful. 
Looking in the logs, it appears the cause is the following:

14:17:48.341 ERROR i.g.d.a.b.b.BroadcastifyCallBroadcaster - Error connecting to Broadcastify calls server on startup [Received close_notify during handshake]  [17MB/38MB 45%]

Any ideas or suggestions on what may be causing this or how to resolve?

DSheirer, 2020-10-12:
@andrewp-sys Something is interrupting the HTTP connection handshake.  Are you using an sdrtrunk release version?  How are you connected to the internet ... are you using WiFi?

andrewp-sys, 2020-10-12:
Issue is resolved in version 0.5.0 alpha 3, I was on alpha 1.

closed 2020-10-12 by andrewp-sys