DSheirer/sdrtrunk #973: RTL2832 Tuners - USB Pipe Error

DSheirer, 2020-10-19:
Investigate if this may be caused by concurrent access to the tuner.  There may be an issue where the two-step write processes might be stepping on each other.

-----------------------------------------

20201013 001738.859 [sdrtrunk pool-1-thread-7] ERROR i.g.d.source.tuner.TunerModel - Error obtaining channel from tuner [RTL2832 SDR/R820T 00001001]  [1GB/1GB 69%]
org.usb4java.LibUsbException: USB error 9: error writing byte buffer: Pipe error
 at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:823)
 at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:803)
 at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.writeI2CRegister(RTL2832TunerController.java:688)
 at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerController.writeR820TRegister(R820TTunerController.java:656)
 at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerController.setPLL(R820TTunerController.java:561)
 at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerController.setTunedFrequency(R820TTunerController.java:193)
 at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:181)
 at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:153)
 at io.github.dsheirer.source.tuner.TunerController.setFrequency(TunerController.java:199)
 at io.github.dsheirer.source.tuner.manager.PolyphaseChannelSourceManager.getSource(PolyphaseChannelSourceManager.java:433)
 at io.github.dsheirer.source.tuner.TunerModel.getSource(TunerModel.java:463)
 at io.github.dsheirer.source.SourceManager.getSource(SourceManager.java:91)
 at io.github.dsheirer.controller.channel.ChannelProcessingManager.startProcessing(ChannelProcessingManager.java:303)
 at io.github.dsheirer.controller.channel.ChannelProcessingManager.startChannelRequest(ChannelProcessingManager.java:261)
 at jdk.internal.reflect.GeneratedMethodAccessor140.invoke(Unknown Source)
 at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
 at java.base/java.lang.reflect.Method.invoke(Unknown Source)
 at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:87)
 at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:144)
 at com.google.common.eventbus.Subscriber$1.run(Subscriber.java:72)
 at com.google.common.util.concurrent.DirectExecutor.execute(DirectExecutor.java:30)
 at com.google.common.eventbus.Subscriber.dispatchEvent(Subscriber.java:67)
 at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:108)
 at com.google.common.eventbus.EventBus.post(EventBus.java:212)
 at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processPhase1ChannelGrant(P25TrafficChannelManager.java:364)
 at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processChannelGrant(P25TrafficChannelManager.java:255)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processChannelGrant(P25P1DecoderState.java:312)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processTSBKGroupVoiceChannelGrant(P25P1DecoderState.java:1912)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processTSBK(P25P1DecoderState.java:1095)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:287)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:148)
 at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:152)
 at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
 at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:96)
 at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:89)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:95)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:37)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.dispatchMessage(P25P1MessageFramer.java:325)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:204)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:70)
 at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:152)
 at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
 at io.github.dsheirer.dsp.psk.PSKDemodulator.broadcast(PSKDemodulator.java:58)
 at io.github.dsheirer.dsp.psk.DQPSKDecisionDirectedDemodulator.calculateSymbol(DQPSKDecisionDirectedDemodulator.java:88)
 at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:115)
 at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:89)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderC4FM.receive(P25P1DecoderC4FM.java:112)
 at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderC4FM.receive(P25P1DecoderC4FM.java:43)
 at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:42)
 at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:23)
 at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
 at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.flush(ReusableComplexBufferAssembler.java:157)
 at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.receive(ReusableComplexBufferAssembler.java:131)
 at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.receive(ReusableComplexBufferAssembler.java:143)
 at io.github.dsheirer.dsp.filter.channelizer.output.TwoChannelOutputProcessor.process(TwoChannelOutputProcessor.java:117)
 at io.github.dsheirer.dsp.filter.channelizer.output.ChannelOutputProcessor.processChannelResults(ChannelOutputProcessor.java:124)
 at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.processSamples(PolyphaseChannelSource.java:248)
 at io.github.dsheirer.source.tuner.channel.TunerChannelSource$ScheduledIntervalProcessor.run(TunerChannelSource.java:315)
 at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
 at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
 at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
 at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
 at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
 at java.base/java.lang.Thread.run(Unknown Source)

20201013 001744.140 [sdrtrunk pool-1-thread-7] ERROR i.g.d.source.tuner.TunerModel - Error obtaining channel from tuner [RTL2832 SDR/R820T 00001001]  [1GB/1GB 72%]
org.usb4java.LibUsbException: USB error 9: read error: Pipe error
 at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.read(RTL2832TunerController.java:860)
 at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.read(RTL2832TunerController.java:886)
 at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerController.getStatusRegister(R820TTunerController.java:626)
 at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerController.isPLLLocked(R820TTunerController.java:595)
 at io.github

closed 2020-10-25 by DSheirer

RE-Specto, 2023-10-22:
just received the same error
sdrtrunk v0.5.3
happens after using the same tuner in sdr# first. sdr# is already closed, sdrtrunk shows the waterfall but trying to play any channel results in 
```
can't play channel
Error: No Tuner Available
```
and in the console
```
2023-10-22 18:30:52.098 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [26MB/56MB 46%]
2023-10-22 18:31:13.884 ERROR i.g.d.s.t.m.PolyphaseChannelSourceManager - Error getting source  [89MB/228MB 39%]
org.usb4java.LibUsbException: USB error 9: error writing byte buffer: Pipe error
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:798)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.writeDemodRegister(RTL2832TunerController.java:683)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.enableI2CRepeater(RTL2832TunerController.java:578)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController$ControllerAdapter.enableI2CRepeater(RTL2832TunerController.java:1507)
        at io.github.dsheirer.source.tuner.rtl.e4k.E4KEmbeddedTuner.setTunedFrequency(E4KEmbeddedTuner.java:284)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.setTunedFrequency(RTL2832TunerController.java:142)
        at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:203)
        at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:175)
        at io.github.dsheirer.source.tuner.TunerController.setFrequency(TunerController.java:244)
        at io.github.dsheirer.source.tuner.manager.PolyphaseChannelSourceManager.getSource(PolyphaseChannelSourceManager.java:454)
        at io.github.dsheirer.source.tuner.manager.TunerManager.getSource(TunerManager.java:501)
        at io.github.dsheirer.source.tuner.manager.TunerManager.getSource(TunerManager.java:427)
        at io.github.dsheirer.controller.channel.ChannelProcessingManager.startProcessing(ChannelProcessingManager.java:382)
        at io.github.dsheirer.controller.channel.ChannelProcessingManager.start(ChannelProcessingManager.java:303)
        at io.github.dsheirer.gui.playlist.channel.ChannelConfigurationEditor.lambda$getPlayButton$3(ChannelConfigurationEditor.java:372)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)
2023-10-22 18:31:13.886 ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [null] - No Tuner Available  [90MB/228MB 39%]
```
back to normal after rebooting the system, but reproduced after using sdr# again

RE-Specto, 2023-10-22:
@DSheirer any ideas? 👆

RonRN18, 2023-10-22:
Have you tried the latest? 0.6.0-Beta2?
Sent from Ron Webb's iPad using Gmail Mobile

RE-Specto, 2023-10-25:
I'll try later on. however I suspect it may be due to USB port instability.

PR #983 (2020-10-25): 973 Tuner Controller Concurrent Locking - Avoid USB Pipe Errors
Resolves #973.  Adds concurrent locking to primary tuner controller methods to avoid concurrent access to USB methods.