DSheirer/sdrtrunk #985: Raspberry Pi 4 port?

WheezyE, 2020-10-25:
Hi, I'm new to SDRTrunk, so sorry if this has been answered before.  I read over some of the discord discussion and issue trackers here, but haven't seen a discussion on porting SDRTrunk to Raspberry Pi 4.  Reading some of the responses to the question in the past, it seems there were some concerns about available resources on Pi 3B+, but I think the Pi4 could handle the resource load.  Java seems port-able too.

Just wondering if anybody has attempted this yet.

whitakerz, 2020-10-25:
@WheezyE i have successfully loaded and used the program on a raspberry pi 4. [https://groups.google.com/g/sdrtrunk/c/RF6LRfJ7MYo](url)

WheezyE, 2020-10-26:
Awesome! Thanks for the leg work. That PDF documenting your steps and commands is wonderful

WheezyE, 2020-10-27:
I see now that compiling Java for sdrtrunk requires a 64-bit operating system and that Raspbian OS is armhf (though there is an experimental arm64 version of Raspberry Pi OS Buster available).  This makes more sense as to why you had to shop around for RPi4 operating systems.  I wonder if sdrtrunk could ever be ported to armhf.

I super appreciate your write-up.  This saves so much time

DSheirer, 2020-10-28:
The limitation for sdrtrunk is the JavaFX libraries have to be compiled for the host OS and architecture.

Bellsoft's Liberica JDK just started shipping support for ARM64 with Java 15 release, which is why I started including an ARM64 linux version of sdrtrunk.

whitakerz, 2020-10-30:
I tested today. Compiling the code on rasberry pi 64 bit OS works too.

lpoels, 2020-11-27:
@whitakerz > WheezyE i have successfully loaded and used the program on a raspberry pi 4. [https://groups.google.com/g/sdrtrunk/c/RF6LRfJ7MYo](url)

Hey man, any chance you have an updated link for the above.
Sorry I'm new to raspberry pi's and struggling with this aswell; Got SDRtrunk working flawlessly on windows and mac but wanted to try a portable. 
I'm Using Ubuntu 64bit ARM on a pi4. installed java 8, 11 openjfx and then tried LibericaJDK with no luck; always get a successful launch all the way up to discovering the USB bus and then this error:

"Loading Library prism_es2 from resource failed: java.lang.UnsatisifiedLinkError: .../15/libprism_es2.so: cannot open shared object file: no such file or directory (possible cause: cant load AMD 64 .so on a AARCH64 platform"

following this error is just a bunch of failed loaded resources seemingly from JavaFX

whitakerz, 2020-11-27:
The information is all the same. In the guide I wrote the link to the
Liberica link takes up two lines though and the PDF version breaks it up
into 2 separate lines. Try this inplace of what is written:

*Java preparation instructions*

sudo apt update

sudo apt upgrade

   1.

   Get the Liberica Java *FULL*SDK 15
   https://bell-sw.com/pages/downloads/#/java-15-current

cd ~/Downloads

mkdir java

cd java

wget
https://download.bell-sw.com/java/15.0.1+9/bellsoft-jdk15.0.1+9-linux-aarch64-full.deb

   1.

   Install the JDK 15

sudo apt install ./bellsoft-jdk15.0.1+9-linux-aarch64-full.deb

   1.

      I ignored the error “Note selection ‘bellsoft-java15-full' instead of
      './bellsoft-jdk15.0.1+9-linux-aarch64-full.deb'' “
      1.

   Verify the installation
   1.

      java -version
      1.

         Will output:

openjdk version "15.0.1" 2020-10-20

OpenJDK Runtime Environment (build 15.0.1+9)

OpenJDK 64-Bit Server VM (build 15.0.1+9, mixed mode)

*!!!!Set JAVA_HOME variable!!!!!*

   1.

   We need to find the JDK folder
   2.

   I am a huge fan of the app “locate ” and this distribution is missing it

sudo apt install mlocate

   1.

   Look for the home folder of the JDK it will look like usr/lib/jvm/JAVA
   VERSION

sudo updatedb

locate jvm

   1.

   in this case it is located /usr/lib/jvm/bellsoft-java15-aarch64
   2.

   set the JAVA_HOME variable in /etc/profile

sudo nano /etc/profile

   1.

   VERY CAREFULLY add the following line to the file RIGHT AFTER THE LAST
   “#” before the IF

export JAVA_HOME="path that you found"

export PATH=$JAVA_HOME/bin:$PATH

   1.

   in this case:

export JAVA_HOME="/usr/lib/jvm/bellsoft-java15-full-aarch64/"

export PATH=$JAVA_HOME/bin:$PATH

whitakerz, 2020-11-27:
Sorry I meant to include the email as well.

lpoels, 2020-11-27:
@whitakerz 
Thanks man! Setting my Java home variable was the issue, makes sense now lol
I appreciate all the time you put into this, cheers!

tim273, 2021-04-16:
One thing that helps a lot is going to View and Disable Spectrum and Waterfall that makes it useable.  Before I did that the audio was choppy even with only one call and the CPU was usually spiked at 100%, but when I shut that off, it's more like 50-75 with two calls.  So it's an improvement.  

I'm working on a Raspberry Pi scanner with a 7 inch screen and Smart Pi case.  It's still in the works, but I'll post more when I'm done.  I've got it working right now with LXDE and Ubuntu Server 20.04.

Backu68, 2021-08-28:
@whitakerz Thanks to your PDF instructions, I have gotten my Pi4b 8GB to run SDR-Trunk, with some limitations.. and using the beta of Raspbian aarch64.. my issue is, with my touchscreen, I cannot change tabs in the Playlist editor so I can start playing streams, or access any of the menus, but I can close it, and operate the main SDR-Trunk window without issue.. any ideas on how to fix the 'touchscreen mouse' clicks on the subs?

whitakerz, 2021-08-29:
It has been a little while since I have had my touch screen out. I have a
generic touchscreen with the WAY generic driver and I found it to be super
limited. I think it is a combination of those factors. I seem to
remember that limitation as well. I don't have the time to reload SDR on my
touch screen to explore right now. Let me know what you find out.

Zach

whitakerz, 2021-08-29:
It might also be a limitation of linux flavor I was using. Maybe someone
who loads it on a different flavor has a different experience. Try the
discord!

WheezyE, 2022-10-12:
Closing this issue as old. RPiOS 64-bit is now easy to install with Raspberry Pi Imager.
Thank you all for your help and support!

closed 2022-10-12 by WheezyE