DSheirer/sdrtrunk #994: ChannelRotationMonitor - Error while checking state

GTR8000, 2020-11-01:
This is a nasty one, which brings decoding to a standstill. Seems to primarily affect P25 systems that have multiple control channels in the channel definition, and it bombs when checking for rotation. This causes the spectrum display to freeze and all decoding stops.

`20201101 115145.218 [sdrtrunk pool-1-thread-2] WARN  i.g.d.s.t.c.r.ChannelRotationMonitor - Error while checking state  [2GB/3GB 68%]
java.lang.IllegalStateException: transferPointer is not initialized
	at org.usb4java.LibUsb.cancelTransfer(Native Method)
	at io.github.dsheirer.source.tuner.usb.USBTransferProcessor.stop(USBTransferProcessor.java:172)
	at io.github.dsheirer.source.tuner.usb.USBTransferProcessor.removeListener(USBTransferProcessor.java:383)
	at io.github.dsheirer.source.tuner.usb.USBTunerController.removeBufferListener(USBTunerController.java:69)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.removeBufferListener(RTL2832TunerController.java:1547)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager.stopChannelSource(PolyphaseChannelManager.java:268)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$ChannelSourceEventListener.receive(PolyphaseChannelManager.java:530)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$ChannelSourceEventListener.receive(PolyphaseChannelManager.java:509)
	at io.github.dsheirer.source.tuner.channel.TunerChannelSource.broadcastProducerSourceEvent(TunerChannelSource.java:214)
	at io.github.dsheirer.source.tuner.channel.TunerChannelSource.stop(TunerChannelSource.java:137)
	at io.github.dsheirer.source.tuner.channel.MultiFrequencyTunerChannelSource.rotate(MultiFrequencyTunerChannelSource.java:95)
	at io.github.dsheirer.source.tuner.channel.MultiFrequencyTunerChannelSource.process(MultiFrequencyTunerChannelSource.java:351)
	at io.github.dsheirer.source.SourceEventListenerToProcessorAdapter.receive(SourceEventListenerToProcessorAdapter.java:51)
	at io.github.dsheirer.source.SourceEventListenerToProcessorAdapter.receive(SourceEventListenerToProcessorAdapter.java:25)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:152)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
	at io.github.dsheirer.source.tuner.channel.rotation.ChannelRotationMonitor.checkState(ChannelRotationMonitor.java:148)
	at io.github.dsheirer.source.tuner.channel.rotation.ChannelRotationMonitor.lambda$start$0(ChannelRotationMonitor.java:179)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)`

GTR8000, 2020-11-02:
This appears to be related. After one of those events, upon attempting to shut down SDRTrunk, it hangs and throws this error:

`Exception in thread "AWT-EventQueue-0" java.lang.IllegalStateException: transferPointer is not initialized
        at org.usb4java.LibUsb.cancelTransfer(Native Method)
        at io.github.dsheirer.source.tuner.usb.USBTransferProcessor.stop(USBTransferProcessor.java:172)
        at io.github.dsheirer.source.tuner.usb.USBTransferProcessor.removeListener(USBTransferProcessor.java:383)
        at io.github.dsheirer.source.tuner.usb.USBTunerController.removeBufferListener(USBTunerController.java:69)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.removeBufferListener(RTL2832TunerController.java:1547)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager.stopChannelSource(PolyphaseChannelManager.java:268)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$ChannelSourceEventListener.receive(PolyphaseChannelManager.java:530)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$ChannelSourceEventListener.receive(PolyphaseChannelManager.java:509)
        at io.github.dsheirer.source.tuner.channel.TunerChannelSource.broadcastProducerSourceEvent(TunerChannelSource.java:214)
        at io.github.dsheirer.source.tuner.channel.TunerChannelSource.stop(TunerChannelSource.java:137)
        at io.github.dsheirer.module.ProcessingChain.stop(ProcessingChain.java:684)
        at io.github.dsheirer.controller.channel.ChannelProcessingManager.stopProcessing(ChannelProcessingManager.java:495)
        at io.github.dsheirer.controller.channel.ChannelProcessingManager.shutdown(ChannelProcessingManager.java:536)
        at io.github.dsheirer.gui.SDRTrunk.processShutdown(SDRTrunk.java:432)
        at io.github.dsheirer.gui.SDRTrunk$ShutdownMonitor.windowClosing(SDRTrunk.java:563)
        at java.desktop/java.awt.Window.processWindowEvent(Unknown Source)
        at java.desktop/javax.swing.JFrame.processWindowEvent(Unknown Source)
        at java.desktop/java.awt.Window.processEvent(Unknown Source)
        at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
        at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)`

DSheirer, 2026-11-07:
This error was in the legacy code.  sdrtrunk is now using a new jRTL driver library.  Closing.  Please reopen if the problem still exists with the new library.

closed 2026-11-07 by DSheirer