DSheirer/sdrtrunk #999: Beep Action Error

DSheirer, 2020-11-18:
7:28:16.182 ERROR i.g.d.a.a.beep.BeepAction - Error attempting to play audio test tone  [393MB/922MB 42%]
17:28:17.139 ERROR i.g.d.a.a.beep.BeepAction - Error attempting to play audio test tone  [407MB/922MB 44%]

FLYBYME, 2022-03-18:
I seem to have the same issue. Its the same with version 0.4. I'm running Ubuntu 20.04

It can seem to pickup they audio device of my system.

If you need more information please let me know.

```
02:58:28.185 INFO  i.g.d.icon.IconModel - loading icons file [/root/SDRTrunk/settings/icons.xml]  [11MB/124MB 9%]
02:58:28.734 INFO  i.g.d.log.ApplicationLog - Application Log File: /root/SDRTrunk/logs/sdrtrunk_app.log  [9MB/27MB 36%]
02:58:28.763 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.5.0-alpha6  [11MB/27MB 42%]
02:58:28.763 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 6.3  [11MB/27MB 42%]
02:58:28.763 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2020-10-26T04:29:57.816-0400  [11MB/27MB 42%]
02:58:28.764 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 15 (BellSoft 15+36  [11MB/27MB 42%]
02:58:28.764 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 4.4.0-193-generic  [11MB/27MB 42%]
02:58:28.764 INFO  i.g.d.log.ApplicationLog -   [11MB/27MB 44%]
02:58:28.764 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [11MB/27MB 44%]
02:58:28.764 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [11MB/27MB 44%]
02:58:28.765 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [11MB/27MB 44%]
02:58:28.765 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [11MB/27MB 44%]
02:58:28.765 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [11MB/27MB 44%]
02:58:28.765 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Linux  [11MB/27MB 44%]
02:58:28.765 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [11MB/27MB 44%]
02:58:28.766 INFO  i.g.d.log.ApplicationLog - Host OS Version:       5.13.0-35-generic  [11MB/27MB 44%]
02:58:28.766 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4  [11MB/27MB 44%]
02:58:28.767 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  1 GB  [11MB/27MB 44%]
02:58:28.767 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [11MB/27MB 44%]
02:58:28.767 INFO  i.g.d.log.ApplicationLog -  Application Root: /root/SDRTrunk  [11MB/27MB 44%]
02:58:28.767 INFO  i.g.d.log.ApplicationLog -  Application Log:  /root/SDRTrunk/logs  [11MB/27MB 44%]
02:58:28.768 INFO  i.g.d.log.ApplicationLog -  Event Log:        /root/SDRTrunk/event_logs  [11MB/27MB 44%]
02:58:28.768 INFO  i.g.d.log.ApplicationLog -  Playlist:         /root/SDRTrunk/playlist  [11MB/27MB 44%]
02:58:28.768 INFO  i.g.d.log.ApplicationLog -  Recordings:       /root/SDRTrunk/recordings  [11MB/27MB 44%]
02:58:28.892 INFO  i.g.d.util.ThreadPool - Application thread pool created with [4] threads  [14MB/27MB 55%]
02:58:28.892 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /root/SDRTrunk  [14MB/27MB 55%]
02:58:28.893 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/root/SDRTrunk/SDRTrunk.properties]  [14MB/27MB 55%]
02:58:28.893 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/root/SDRTrunk/SDRTrunk.properties]  [14MB/27MB 55%]
02:58:28.905 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/root/SDRTrunk/settings/settings.xml]  [15MB/27MB 59%]
02:58:28.963 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [17MB/27MB 66%]
02:58:28.976 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478  [18MB/27MB 68%]
02:58:28.977 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312  [18MB/27MB 68%]
02:58:28.977 INFO  i.g.d.s.t.TunerManager - discovered [7] attached USB devices  [18MB/27MB 68%]
02:58:28.982 INFO  i.g.d.s.t.TunerManager - USB Bus [4] Device [1D6B:0003] Hub Device  [18MB/27MB 68%]
02:58:30.505 INFO  i.g.d.s.t.TunerManager - USB Bus [3] Device [0BDA:2838] LOADED: RTL2832 SDR/R820T 00000001 Max Rate:38400000 bps  [7MB/27MB 26%]
02:58:30.506 INFO  i.g.d.s.t.TunerManager - USB Bus [3] Device [1D6B:0002] Hub Device  [7MB/27MB 26%]
02:58:30.507 INFO  i.g.d.s.t.TunerManager - USB Bus [2] Device [1D6B:0003] Hub Device  [7MB/27MB 26%]
02:58:30.508 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [046D:C077] Unknown Device - Class 0  [7MB/27MB 26%]
02:58:30.509 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [046D:C31C] Unknown Device - Class 0  [7MB/27MB 27%]
02:58:30.510 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [1D6B:0002] Hub Device  [7MB/27MB 27%]
02:58:30.511 INFO  i.g.d.s.t.TunerManager - -------------------------------------------------------------  [7MB/27MB 27%]
02:58:30.511 INFO  i.g.d.s.t.TunerManager - USB Bus - Potential Maximum Data Rates  [7MB/27MB 27%]
02:58:30.518 INFO  i.g.d.s.t.TunerManager - USB Bus [3] Rate [38400000] bits per second  [7MB/27MB 28%]
02:58:31.052 ERROR i.g.d.a.p.AudioOutput - Couldn't obtain audio source data line for audio output - mixer [HDMI [plughw:0,9]]  [16MB/27MB 59%]
```

DSheirer, 2022-03-19:
@FLYBYME It looks like you need to select another audio output.  Java is selecting your HDMI audio output by default and sdrtrunk can't output audio to that device.

You can change the audio output device in the User Preferences dialog.  You may be able to select among several audio output devices.  Using the 'Test' button until you find the correct one.

FLYBYME, 2022-03-28:
Every interface you select you get the same error.

I believe the issue is I was running sdrtrunk as root. I got audio to work without sudo, but not with sudo.