DSheirer/sdrtrunk #1002: Create checkbox to disable auto-tuning

tvsjr, 2020-11-30:
I am running an Airspy to track several purely 700MHz systems. I experience sync loss when new traffic channels are created, as the software attempts to pick the "most optimum" center frequency and retunes the tuner, causing a short drop. This drop is usually enough to send the software on a search to reidentify the control channel, which takes a bit on a Harris system where every channel is possibly a control channel.

There are other issues open for extensive overhaul of the tuner management, but my wish is simple - include a check box to prevent the tuner from being auto-tuned. This would allow me to set the Airspy at a good center frequency (say, 772.0000MHz) which would cover the entire 700 spectrum. If a voice channel were to come along outside the range (say, an 800 channel on a dual-band system), the software would simply not tune the channel due to resource limitation.

GTR8000, 2020-11-30:
#904

davidflypei, 2024-07-26:
+1 on this. I'm currently running into an issue where it keeps tuning things to the right, and it pushed one of the channels so far to the right that if it does stay triggered, it is super quiet, but most of the time, it chops things up, and I can't get anything out of it that is useful. There's tons of space to the left for it to move over (purple), but it insists on shoving the channel circled in red to the left into the bandpass cutoff fade.

[image]

tvsjr, 2024-07-26:
@davidflypei as a workaround, you can create two NBFM receivers to "pin" the receiver. For example, on an Airspy monitoring 700, I create "Airspy #1 Low" at 767.50625 and "Airspy #1 High" at 776.49375 and set both to the same preferred tuner. I also set them to a start order of 1, with everything else at 2 or below, so they will start and pin the frequencies before anything else has an opportunity to conflict.

That particular system has a total of 5 Airspys running on it, covering VHF, UHF, 700, and 800. I ensure there are no overlaps between any of the tuners (which are all pinned in this manner, so no moving around), then leave all the other decoders with no preferred tuner. Since there's always no more than one tuner responsive to a specific frequency, it works great.