DSheirer/sdrtrunk #1003: Feature request: Allow single-dongle operation by tuning away from the control channel temporarily

spsx, 2020-12-01:
For users with single dongles, the bandwidth of the SDR may not be enough to pick up the control channel as well as the voice channel simutaneously. This causes a "Tuner unavailable" error to appear in the "Events" box, and the voice channel is not picked up.

E.g. I have a control channel at 400MHz, Control indicates that a voice call is occuring at 403MHz. Since my SDR only has 2MHz bandwidth, I get the "Tuner unavailable" error.

I'm not sure about other SDR models, but the RTL-SDR can retune itself very quickly. **Therefore it would be nice if sdrtrunk would have the option to temporarily tune the SDR off the control channel to listen to the voice channel, then switch back to monitoring the control channel**. Obviously this would mean that there is a chance that some control data would be lost, but it would be better than not picking up anything at all.

Thanks!

hayden-t, 2020-12-23:
apparently the control data is in the  voice streams as well, so that client units dont need wide band or dual band tuners, and can follow control while on voice. op25 apparently does this switching

DSheirer, 2026-11-07:
This will not be implemented.

closed 2026-11-07 by DSheirer