DSheirer/sdrtrunk #1007: Feature Request: Zello Integration

josephdambrosio, 2020-12-03:
I use SDR trunk in combination with Zello's offer for free accounts for first responders to connect our dept to a listen only version of our county's P25 TGs.  Right now, zello keys up through VOX which doesn't work all the time.  It supports serial connections to activate the PTT and it would be cool of SDR Trunk supported some type of "virtual serial" connection to keyup zello.

taz420nj, 2020-12-08:
I don't even think there needs to be serial, although it would probably prove to be useful for other applications.  If SDRT could output certain TGs or groups of TGs to a specific output (via VAC) as it does with Icecast, as well as allow (under Actions) it to activate a hotkey combination (which could be set differently to each instance of Zello) when those specific TGs are active - and of course allow multiple hotkeys to be activated simultaneously if needed, it should work to route specific TGs to specific channels..

taz420nj, 2020-12-08:
I don't even think there needs to be serial, although it would probably prove to be useful for other applications. If SDRT could output certain TGs or groups of TGs to a specific output (via VAC) as it does with Icecast, as well as allow (under Actions) it to activate a hotkey combination (which could be set differently to PTT each instance of Zello) when those specific TGs are active - and of course allow multiple hotkeys to be activated simultaneously if needed, it should work to route specific TGs to specific channels..

But I am also looking to set up a private streaming system via Zello and would highly appreciate being able to separate out certain TGs to specifc Zello channels.  I can't help with the coding but I'm more than willing to help test it out.

LordDonkey, 2020-12-27:
It's already on the list.
https://github.com/DSheirer/sdrtrunk/issues/828