DSheirer/sdrtrunk #1010: NO SOURD

gaspar1987, 2020-12-11:
[image]

gaspar1987, 2020-12-18:
Bonjour monsieur je voudrais savoir comment ça se fait que je n'ai pas de son sur mon décrypteur je suis sur windows je n'arrive pas à décrypter avec votre logiciel sdrtunk merci de me donner la réponse pour que je puisse réparer où mettre des paramètres qu'il manque dans votre logiciel expliquez-moi le avec une photo ou quelque chose qui pourrait me montrer comment on peut utiliser sans que je puisse avoir des erreurs quand je le manipule le logiciel est très bien constituer mais je n'arrive pas à mettre en route le son de l'appareil en lecteur play.

trengking, 2020-12-18:
it is easier for others if you would spend a little more time to translate it in english.

gaspar1987, 2020-12-18:
Hello, sir, I used your software there is a problem, I can not listen to the sound of the dr

gaspar1987, 2020-12-18:
Hello sir, I can't see the DR ds that works on audio mode

gaspar1987, 2020-12-19:
Hello I would like to know how you do to program the library to have the sound DMR thanks for giving me to create the library I do not understand how to program it you can give me a tip with photos or a better instructions because I do not understand how it is made
[image]

LordDonkey, 2020-12-19:
You have no JMBE library set. Click on "Create Library." It will automatically download and compile the latest JMBE library.

https://user-images.githubusercontent.com/52051804/102694219-a2d73700-41ed-11eb-9be6-1c6b94935a30.mp4

gaspar1987, 2020-12-19:
[](

gaspar1987, 2020-12-19:
IMG_20201219_173955_8.jpg -->

gaspar1987, 2020-12-19:
Hello, sir, you're using this with the internet because it doesn't work. I tested it and it doesn't do the same thing you reproduced on the video

gaspar1987, 2020-12-19:
Hello, I would like you to contact him to put the bookstore on it, because I don't and

gaspar1987, 2020-12-19:
Hello, sir, explain to me how we can get the sound with

closed 2020-12-19 by gaspar1987

gaspar1987, 2020-12-26:
Good morning.
I've set up.
Your software, but it doesn't work. I don't have the.
His.
Can you give me any indications that allows?
From.
Have the audio by showing me how it can be installed? Already because I have problems, the sound does not work. I created the library, it doesn't work.