DSheirer/sdrtrunk #1017: UHD driver from Ettus Research for USRP

UR21F, 2020-12-19:
I am very grateful for your work! Do you plan to support USRP B210?

maozhenyu123, 2020-12-20:
I might want to do this. 

First step, get one USRP.

jnse, 2021-06-16:
Might i suggest an alternate way to do this, that doesn't depend on getting an USRP? 
Most USRP's work fine with the osmosdr driver, which also supports cheap rtl dongles afaik.
So if you implement support for osmosdr using a generic sdr, it will likely also work for USRP devices.

The only important thing is that you let the user pass arbitrary device strings for initialization.
(eg: type=uhd,addr=192.168.1.200,clock_source=external,ppm=1.0   - or whatever else people might want to initialize their usrp with)

sherdil6, 2023-04-19:
Any success using USRP  with SDRTrunk? I think it is still pending. I need this for B205 mini.

Josh-SMU, 2023-09-01:
Bumping, I cannot use my B205 mini with SDRtrunk

DominicWeston, 2024-05-12:
Bump, would love to use my big old white PCB B200.

PowerPress, 2024-12-05:
Any updates on this?