DSheirer/sdrtrunk #1027: Add Support for OpenMHS

Pcpro745, 2021-01-05:
Not sure if this is possible but can you add support for uploading to OpenMHZ?

ImagoTrigger, 2021-06-01:
is there any documentation for this or is it in trunk-recorder somewhere?

Pcpro745, 2021-07-12:
its in Trunk Recorder

Pcpro745, 2023-01-11:
Bumping this again

theficus, 2023-11-09:
Adding a +1 to this. Right now I have to use a gnarly combination of running a Hyper-V Linux VM running trunk-recorder and proxying my RTL-SDRs via usbip so I can upload to both Broadcastify and OpenMHz simultaneously. Being able to do this through SDRTrunk would save a lot of layers of complexity and frustration in my setup.

tadscottsmith, 2023-11-10:
@DSheirer are you open to an OpenMHz broadcaster? I could tackle that.

DSheirer, 2023-11-10:
Sure!

Ctpolicelive, 2023-12-17:
Working perfect. Great feature. Thanks.

jmp367, 2024-01-09:
When using 0.6.0 to feed OpenMHz, the feed periodically becomes disconnected and will not reconnect to OpenMHz automatically. SDRTrunk shows "Error" for the OpenMHz connection status when this occurs and there are no messages related to this issue in the log file.

DSheirer, 2024-01-10:
@jmp367 this issue should have been closed because the scope for this issue was to implement the original feature.  If there's a bug in the current implementation that you're reporting, please open a new issue.

closed 2024-01-10 by DSheirer