DSheirer/sdrtrunk #1029: Add  support for Fitipower tuners

covex, 2021-01-09:
In the https://github.com/DSheirer/sdrtrunk/issues/546 it was meantioned that you may add a support for FC0012/13 tuners that are common compannions for RTL2832/8 chips, but it is probably still not there as they do not work with SDRTrunk. They do work OK with all other SDR softwares.

iComputer7, 2023-08-24:
Still seems unsupported as of v0.6.0-alpha7. Last mention seems to be #1474 in March 2023

Xeon64, 2023-11-30:
I've noticed you just added support for FC0013 tuner based on the commit messages. I see some additions for the FC0012 tuner but it still doesn't seem to work. Any idea when support for the FC0012 tuner will be fully added? After taking a quick look at the tuner_fc0012.c + tuner_fc0013.c from rtl_fm there doesn't seem to be a huge difference between the two.  If FC0013 is already working with sdr-trunk I wouldn't think FC0012 would be hard to add even if it was commented out for testing purposes only.

-Thanks

I bought like 20+ of these generic devices before I knew it had the FC0012 tuner or it's limitations. Although I still have a couple old/original RTL devices that still work with most things.. I would like to get these FC0012 tuners working with SDR-TRUNK to see what large group of RTL devices can do.

Bus 002 Device 015: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

2023-11-30 16:42:16.864 INFO  i.g.d.s.t.m.DiscoveredTuner - Tuner Error - Stopping - RTL-2832 USB Bus:2 Port:12.4.1.1.1 Error: Unsupported/Unrecognized Tuner Ty
pe: FC0012  [80MB/2GB 3%]

jsonpoindexter, 2024-04-27:
@Xeon64 thank you for this!

Xeon64, 2024-05-14:
> @Xeon64 thank you for this!

You're welcome.. I plan on getting back to this project in the next month or so.. I need to update my cheap antennas to a single split antenna.. before I start running the hardware for testing again.. too many wires.  I was also thinking about making a custom PCB with a built-in usb hub with all the chips on the board it would make things nice and neat.. and get rid of all the single usb devices in a hub.. and I can skip making custom wires/boosters/filters.. to clean things up and do it right on the PCB board.

L8rGator, 2025-09-03:
> > [@Xeon64](https://github.com/Xeon64) thank you for this!
> 
> You're welcome.. I plan on getting back to this project in the next month or so.. I need to update my cheap antennas to a single split antenna.. before I start running the hardware for testing again.. too many wires. I was also thinking about making a custom PCB with a built-in usb hub with all the chips on the board it would make things nice and neat.. and get rid of all the single usb devices in a hub.. and I can skip making custom wires/boosters/filters.. to clean things up and do it right on the PCB board.
> 
> In my last test I did notice a bug on systems with more then one GPU.. audio output did not work when there is more than 1 GPU on the system.. no matter which hw port sdrtrunk was set to.. it wouldn't work or list the correct port used by the main audio GPU. I'll work on that issue too when I get a chance. It didn't seem to have anything to do with my code that I updated or any unusual system configs other then having more than one GPU.

> I've noticed you just added support for FC0013 tuner based on the commit messages. I see some additions for the FC0012 tuner but it still doesn't seem to work. Any idea when support for the FC0012 tuner will be fully added? After taking a quick look at the tuner_fc0012.c + tuner_fc0013.c from rtl_fm there doesn't seem to be a huge difference between the two. If FC0013 is already working with sdr-trunk I wouldn't think FC0012 would be hard to add even if it was commented out for testing purposes only.
> 
> -Thanks
> 
> I bought like 20+ of these generic devices before I knew it had the FC0012 tuner or it's limitations. Although I still have a couple old/original RTL devices that still work with most things.. I would like to get these FC0012 tuners working with SDR-TRUNK to see what large group of RTL devices can do.
> 
> Bus 002 Device 015: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
> 
> 2023-11-30 16:42:16.864 INFO i.g.d.s.t.m.DiscoveredTuner - Tuner Error - Stopping - RTL-2832 USB Bus:2 Port:12.4.1.1.1 Error: Unsupported/Unrecognized Tuner Ty pe: FC0012 [80MB/2GB 3%]
> 
> UPDATE: 12/01/2023 I did get 10+ FC0012 tuners to work with SDR-TRUNK however I'm not getting the same output I do with gqrx/osmosdr.. so it doesn't tune to any correct channels or at least pick them up even with the correct PPM set that I use in gqrx.. I clearly don't have everything set correctly with the driver to correctly tune into the 700-900mhz band like gqrx does. Frequencies don't pickup like they do in gqrx and center frequency seems way off than the normal PPM you find in gqrx.
> 
> UPDATE: 12/02/2023 After updating the correct frequencies and registers in the driver all is working great ! I'll try to make a fork with all the updates when I have a chance. I currently have 7 FC0012 SDRs working fine when I get more than 10+ SDRs I get a USB error -9 on all the SDRs.. I suspect it's power issue or a USB flooded bus.. I have 4 separate USB controllers on my system... I was only using one of them. I just need more hubs to add more DC amps and switch them off to separate controllers on the system. I should be able to get all 20+ FC0012 SDRs that I have working without a issue on a single SDR-TRUNK process.
> 
> I'll be sure to add binaries for those who don't know how to compile from source.
> 
> UPDATE: 12/02/2023
> 
> The source and binaries for (Linux/OSx aarch_64 + x86_64 + Windows_64) with the updated drivers can be found at my fork : [sdrtrunk-FITIPOWER-FC0012](https://github.com/Xeon64/sdrtrunk-FITIPOWER-FC0012)
> 
> I'll clean up some more of the code later in future beta releases and possibly add extra decoders for various streams.

I got this fork to work great for listening.  But it refused to record anything or stream to openmhz (and I think broadcastify, but it's been a while since I worked on it).  Was this part working for you?  I ended up buying supported dongles for that project, but I'd like to get these FC0012's working on a different system now.

DSheirer, 2026-11-07:
Closing as implemented.

closed 2026-11-07 by DSheirer