DSheirer/sdrtrunk #1031: Talkgroup hold or selection

Motech1949, 2021-01-20:
Downloaded this SDR TRUNK latest Beta, works fantastic far superior to my older RS Pro-96, which became obselete when local system moved to 700 Mhz. Pro-96 would often garble calls or just output raw data. 
What would be nice is to have 'talkgroup hold' or the ability to listen to maybe 1-3 talkgroups by typing in their IDs. System works great but sometimes is like getting a drink out of a firehose LOL. (and I'm only using 1 dongle at the moment). Thanks again for a great piece of software.
Regards, Rich

edmk5000, 2021-03-04:
Agreed. A busy system has continuous talkgroup activity and there is no way to select or hold just one talkgroup. Whitelisting/Blacklisting 100 groups is impossible to do in time when an important talkgroup pops up.

I cant believe sdrtrunk is so good but does not include aA PAUSE, HOLD, or talkgroup selection input would make sdrtrunk

zerog2k, 2021-07-22:
dupe of #50