DSheirer/sdrtrunk #1033: Receiver center frequency hopping

ah1102, 2021-01-24:
I have a hackrf receiver tuned to 6 MHz bandwidth. This is enough to control the range of interest in P25 phase 1. But I see that the center frequency of the receiver is constantly jumping and because of this there are short interruptions in the reception of signals. Is it possible to somehow solve this problem, for example, add a button to lock the center frequency of the receiver?

ah1102, 2021-01-26:
This happens if the program is tracking two sites. And if 1 site gives out channel frequencies below its frequency, and the other above.

jamren01, 2021-01-27:
Add another channel system entry with the low of the range that you want to lock the receiver to. Set it to auto start with a priority higher than your control channels.

ah1102, 2021-01-27:
Thank you) I'll try. I completely forgot about channel priorities.

ah1102, 2021-01-27:
jamren01, do you know what could be the problem? When accepting three sites, one of them periodically falls off (idle). Its signal strength is excellent all the time. ppm is compensated and the tuning frequency is accurate.

closed 2023-03-21 by DSheirer