DSheirer/sdrtrunk #1035: P25-1 failures when adding a 3rd site

ah1102, 2021-01-31:
When adding a third site to monitor the P25-1 network, the newly added site fails intermittently. Separately, the site works smoothly.

-Thermal compensation enabled -0.6Hz for my device

What could be the reason, how to figure it out?

```
 decoded_messages.log

20210131 131150,PASSED,NAC:716 TDU  
20210131 131150,PASSED,NAC:716 AMBTC NET_STATUS_BCAST WACN:781824 SYSTEM:706 CHAN:2-1660/2-3208 SERVICES:[DATA, VOICE, REGISTRATION]
20210131 131150,PASSED,NAC:716 TDU  
20210131 131150,PASSED,NAC:716 AMBTC ADJ SITE STATUS  SYSTEM:706 LRA:0 RFSS:1 SITE:5 CHAN:2-1696/2-3218
20210131 131150,PASSED,NAC:716 TDU  
20210131 131150,PASSED,NAC:716 TSBK1 IDEN_UPDATE_VUHF ID:2 OFFSET:-6400000 SPACING:6250 BASE:440000000 FDMA BW:12500
20210131 131150,PASSED,NAC:716 TSBK2 GRP_VCH_GRNT_UPD GROUP A:60 CHAN:2-1924 GROUP B:1320 CHAN:2-1832
20210131 131150,PASSED,NAC:716 TSBK3 GRP_VCH_GRNT_UPD GROUP A:1360 CHAN:2-1764 GROUP B:117 CHAN:2-1712
20210131 131150,PASSED,NAC:716 AMBTC GRP_VCH_GRANT    FM:136435 TO:1360 CHAN:2-1764/2-3364 SERVICE OPTIONS:PRI4 CIRCUIT
20210131 131150,PASSED,NAC:716 TDU  
20210131 131150,PASSED,NAC:716 AMBTC GRP_VCH_GRANT    FM:136435 TO:1360 CHAN:2-1764/2-3364 SERVICE OPTIONS:PRI4 CIRCUIT
20210131 131150,PASSED,NAC:716 TDU  
20210131 131150,FAILED,NAC:716 TSBK1**CRC-FAILED** VENDOR:V3 UNRECOGNIZED OSP OPCODE MSG:020380D4513B50A007F7757E
20210131 131150,FAILED,NAC:716 TSBK2**CRC-FAILED** VENDOR:V255 UNRECOGNIZED OSP OPCODE MSG:27FFD135EFFFF92B10AF3D11
20210131 131150,FAILED,NAC:716 TSBK3**CRC-FAILED** VENDOR:V100 UNRECOGNIZED OSP OPCODE MSG:56643D5FFFFFFFFFFFFF68F8
20210131 131150,FAILED,NAC:716 TSBK3**CRC-FAILED** VENDOR:V42 UNRECOGNIZED OSP OPCODE MSG:D72A7C70D44184FF3D24F445
20210131 131150,PASSED,<-> SYNC LOSS - BITS PROCESSED [524]
20210131 131150,FAILED,NAC:716 TSBK1**CRC-FAILED** VENDOR:V2 UNRECOGNIZED OSP OPCODE MSG:3D02FD3F4040004000015F58
20210131 131150,FAILED,NAC:716 TSBK2**CRC-FAILED** VENDOR:V5 UNRECOGNIZED OSP OPCODE MSG:80059E0000000000000A7F74
20210131 131151,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131151,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131152,PASSED,<-> SYNC LOSS - BITS PROCESSED [9600]
20210131 131152,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131152,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131153,PASSED,<-> SYNC LOSS - BITS PROCESSED [9600]
20210131 131152,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131153,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131153,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131153,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131154,PASSED,<-> SYNC LOSS - BITS PROCESSED [9600]
20210131 131154,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131154,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131155,PASSED,<-> SYNC LOSS - BITS PROCESSED [9600]
20210131 131154,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131156,PASSED,<-> SYNC LOSS - BITS PROCESSED [9600]
20210131 131155,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131156,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131156,PASSED,<-> SYNC LOSS - BITS PROCESSED [6334]
20210131 131156,PASSED,NAC:716 TDU  
20210131 131156,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210131 131156,PASSED,<-> SYNC LOSS - BITS PROCESSED [2172]
20210131 131156,PASSED,NAC:716 TSBK1 IDEN_UPDATE_VUHF ID:4 OFFSET:-6400000 SPACING:6250 BASE:160587500 FDMA BW:12500
20210131 131156,PASSED,NAC:716 TSBK2 GRP_VCH_GRNT_UPD GROUP A:1302 CHAN:2-1860 GROUP B:1201 CHAN:2-1988
20210131 131156,PASSED,NAC:716 TSBK3 GRP_VCH_GRNT_UPD GROUP A:113 CHAN:2-2056 GROUP B:1415 CHAN:2-1792
20210131 131156,PASSED,NAC:716 AMBTC RFSS_STATUS_BCST SYSTEM:706 SITE:12 ACTIVE NETWORK CONNECTION RFSS:2 LRA:0 CHAN:2-1660/2-3208
20210131 131156,PASSED,NAC:716 TDU  
20210131 131156,PASSED,NAC:716 AMBTC GRP_VCH_GRANT    FM:1011017 TO:117 CHAN:2-1712/2-3312 SERVICE OPTIONS:PRI4 CIRCUIT
20210131 131156,PASSED,NAC:716 TDU  
20210131 131156,PASSED,NAC:716 AMBTC GRP_VCH_GRANT    FM:1011017 TO:117 CHAN:2-1712/2-3312 SERVICE OPTIONS:PRI4 CIRCUIT
20210131 131156,PASSED,NAC:716 TDU  
20210131 131156,PASSED,NAC:716 TSBK1 GRP_VCH_GRNT_UPD GROUP A:1201 CHAN:2-1988 GROUP B:117 CHAN:2-1712
20210131 131156,PASSED,NAC:716 TSBK2 IDEN_UPDATE_VUHF ID:2 OFFSET:-6400000 SPACING:6250 BASE:440000000 FDMA BW:12500
20210131 131156,PASSED,NAC:716 TSBK3 SYSTEM LOADING   MOTOROLA MSG:89901400000000000000E050
20210131 131156,PASSED,NAC:716 AMBTC GRP_VCH_GRANT    FM:0 TO:113 CHAN:2-2056/2-3656 SERVICE OPTIONS:PRI4 CIRCUIT
20210131 131156,PASSED,NAC:716 TDU  
```

CPU load, in my opinion, is not extreme

[image]
[image]
[image]

David-Forg, 2021-03-16:
Have you got enough bandwidth on your SDR to cover all 3 control channels?

ah1102, 2021-03-21:
Yes, hackrf can set channel width up to 20 MHz

ah1102, 2021-03-21:
There is another problem associated with this, that when monitoring 3 sites, the central frequency of the device can change for any channel width. This, as I noticed, can lead to errors in signal reception. The program does not block the frequency of the receiver, or at least check if the channel width is sufficient so as not to make unnecessary tuning.