DSheirer/sdrtrunk #1042: ppm adjustment may be off by factor of 4

ac2cz, 2021-02-14:
This is likely minor so feel free to ignore or close.  

I notice that the ppm adjustment routine in RTL2832TunerController calculates the value to write like this:
        int offset = -ppm * TWO_TO_22_POWER / 1000000;

In the librtl code the adjustment uses:
       int16_t offs = ppm * (-1) * TWO_POW(24) / 1000000;

So it is 4 times higher.  The only effect is that it adjusts more slowly, but also if someone calibrates on one program then the ppm value will need to be 4 times higher in sdrTrunk.  Unless I missed something.

GTR8000, 2021-02-15:
Are you basing this off of actual experience setting a dongle's PPM in SDRTrunk, or solely on the formula you're looking at in the code? I have a few installs of SDRTrunk running with various types of dongles requiring corrections ranging from -2.5 to +175, and I have never had to vary the PPM from whatever I would use in Unitrunker or DSD+ or other software for those same dongles. If anything, I've found SDRTrunk to be a bit more accurate with very responsive auto-correction.

ac2cz, 2021-02-15:
Yes, I am basing it off the formula.  I use similar code in another project and noticed the difference when comparing to the librtl c source code that was originally ported for sdrTrunk.  So I raised it just because I noticed it.  As I said.  Feel free to ignore and close if its not an issue.