DSheirer/sdrtrunk #1043: V.0.5.0-alpha6 Vis C++ Runtime Library Assertion failed

Swarfmaker, 2021-02-14:
I hope this is the right place to report this.

Microsoft Visual C++ Runtime Library
Assertion failed
Program: libusb4java.dll
File: ..\libusb\os\poll_windows.c
Line: 145
Expression: fd != NULL

Using 2 RTL-SDR.com dongles, and many entries in the alias list.

Swarfmaker, 2021-02-16:
I found that this error is associated with a flaky USB connection of the SDR dongle.

So it looks like the issue is that the program doesn't recover gracefully from loss of communication with the dongle.

owolter, 2021-06-21:
Same error here, almost daily at random times. 

@Swarfmaker Did you ever resolve this?  If so, do you have any USB cable/hub brand recommendations?

I have two dongles connected to an unpowered hub that seem to be having this flaky USB connection issue. I thought it was movement related, despite the whole setup being in a spot where they aren't touched, so I cable-tied them to the hub to stop movement, but it still happens.

They're not truly disconnecting in Windows, so restarting sdrtrunk gets things going again. The problem is sdrtrunk should be recovering when this happens, and there doesn't seem to be a way to detect this happening and restart the program without actively monitoring for this problem.

Swarfmaker, 2021-06-22:
No, I never resolved it. I was able to make it happen when I used a certain USB port on my computer and I physically disturbed the SDR dongle. I assumed it was a software problem, as you point out.

boteman, 2021-06-22:
As Joanne Dow has pointed out many times on other SDR lists, the culprit is almost always bad USB hub hardware or maybe USB drivers. There's a lot of cheap junk being sold out there, even on "quality" motherboards.

Keep trying different ones until you find one that's reliable.

closed 2026-11-07 by DSheirer