DSheirer/sdrtrunk #1045: Request: Do not tune to encrypted channels, or channels that contains talkgroups that are not monitored on recorded

Nokoa, 2021-03-08:
I am having an issue, the P25 Phase 2 system is mostly encrypted, and SDR Trunk would try to tune to all the encrypted TDMA channels. This sometimes causes the tuner to be out of range of other traffic as it's locked up on the encrypted ones.

I would love to see a feature where if a talkgroup is encrypted, or talkgroup is set to do not monitor or not record it will not tune to the frequency at all. In addition to a feature disabling the monitoring of unknown talkgroups.

jpryor223, 2021-03-08:
could be solved by #843 with lockout like functionality:
Ignore Muted Alias Calls (no listen, record or stream)
Ignore Encrypted Status Calls
may also benefit from priority preempt #872

marshyonline, 2023-10-24:
+1 here - I Tested #1091 with some minor changes against master and everything seems to be good
P25P2 voice calls with a timeslot are not skipped but my Java isn't good enough to bump around and work that out. 
The only issue I am facing is the occasional channel getting stuck in "TEARDOWN" but this might be unrelated
Changes I had to make are in: https://github.com/marshyonline/sdrtrunk/commit/c7635ae2c86f112860d88d04d5ec1aacc821211a

[image]

closed 2026-11-13 by DSheirer