DSheirer/sdrtrunk #1052: Login to RadioReference fails via Playlist. - Login Failed - Please Verify username and password

wa3riz, 2021-04-12:
I have a RR premium account and I've changed the password to remove special characters.

SDRTrunk version v0.5.0-aplha6
Windows 10

I am unable to access the radioreference db with my premium account. It is failing on credentials.
If I try to use "Test Connection" I get "Login Failed  Please verify username and password."  and this error message appears in the log after acknowledging the error wit "OK".
"ERROR i.g.d.g.p.r.LoginDialog - Login failed. Invalid username or password.  Can't login to radioreference.com"

If I just try to login, after hitting "OK", nothing happens and there is no ERROR message in log.

Thank you for the work you do on this application.  I find the P25 decoding to be much better than DSD+ Fastlane.  It appears to be a very versatile app but before I dig into it I'm stuck on this issue .

keywords: radioreference.com, sdrtrunk, playlist, login, RRDB, RR db, unable

wa3riz, 2021-04-14:
Since I cannot get logged into radioreference how to I manually import my county's P25 data??
It seems like there ought to be an import method for loading a CSV file.

hans109h, 2021-04-14:
Check out the Goggle group for the project, it's been discussed there a number of times and you'll find a python script that can help you with the import.  You might need to tweak it as I don't recall what version of the playlist file it was designed for.

wa3riz, 2021-04-16:
hans109h

Thank you, but I wouldn't say it has been discussed a number of times on the group.  I could find a couple references which dated back to 2017.  They did include a Perl script and I got that to create what looks like a group of aliases but when I try to add them to the Playlist, the app zeros out the entire file.  Which I know means the app sees the file as improper in some way.

I'd like to know what the Playlist format is.  I could only find an old template.

If this conversation is to continue, should we move it to the Google group?

Jeff

wa3riz, 2021-04-24:
Issue Resolved: Radioreference issue with password management resolved by RR by forcing a manual reset of password.