DSheirer/sdrtrunk #1055: Router/modem reboot issue

ChrisABQ, 2021-04-25:
I recently had a new issue with a window 10 computer running version 20H2.  I can run SDRTrunk on the machine, but when I go to stream to Broadcastify, after anywhere between one to three minutes, it triggers the router/modem to reboot (which takes 10 minutes).  I'm able to stream on a windows 10 2004 version machine with no issues, but cannot figure out why the 20H2 machine is causing this issue.  I exchanged the modem, tried different drivers and updates from MS and Intel, nothing seems to work.  All information is pointing to the updated windows version.  Looking for possible solutions.  Note:  This only happens when enabling the stream function.

boteman, 2021-04-25:
What make and model router? 

Consumer-grade equipment can suffer some strange problems, like opening more sessions than it can handle.

ChrisABQ, 2021-04-25:
Thanks for the quick reply.  Was a weird issue but resolved it by changing
the port address from the imported port of 80 to 8500.
One more question, since I have your ear, is there a way to change the
format of the Alpha Tag output to just show the Talkgroup name only?
Thanks

boteman, 2021-04-26:
No idea. I’ve never used SDRtrunk, it’s simply on my ever-growing to do list. We’ll probably be off radio and onto mental telepathy by the time I get around to it :-)

 

 

Bote Man

 <http://www.botecomm.com/bote/radio/streaming.html> http://www.botecomm.com/bote/radio/streaming.html

 

Chris@Dynodan.com <mailto:Chris@Dynodan.com>

GTR8000, 2021-04-28:
"Is there a way to change the format of the Alpha Tag output to just show the Talkgroup name only?"

No, it's hard coded that way. I posted the same answer in your thread on RR, but you never acknowledged it. 🤷🏼‍♂️

closed 2022-04-24 by DSheirer