DSheirer/sdrtrunk #1061: QUESTION: Are these Suggestions supposed to be posted on the RR SDRTrunk Forum???

wa3riz, 2021-05-01:
Are Suggestions supposed to be posted on the RR SDRTrunk Forum???

If so - my apologies for messing up your issues log.  I'll move them asap.

wa3riz, 2021-05-01:
Or the google group?

GTR8000, 2021-05-01:
There is no dedicated SDRTrunk forum on RR. Posting here is just fine, there are plenty of feature requests already posted. The SDRTrunk Google group is fine also. Just understand that the developer has been very busy with his real job over the past few months, and so SDRTrunk is on the back burner for the time being.

wa3riz, 2021-05-02:
That I totally I get.  If I thought I could get help with the coding, I'd role up my sleeves.
It's great that he or she has someone like you running cover for them.

GTR8000, 2021-05-02:
I'm no one special, just another end-user who happens to have chatted with the developer over the years and is aware of his current situation with work. Better to make people aware of that than have them think he has abandoned the project entirely.

closed 2026-11-06 by DSheirer