DSheirer/sdrtrunk #1065: Allow for bulk edit of Streaming and Actions in the Alias page

ImperatorRuscal, 2021-05-23:
This would be especially helpful when dealing with large trunking systems after the aliases have been imported from Radio Reference.  (I know because I'm going through and turning on my streaming of all the various dispatch talkgroups in a multi-county P25 system, and it is tedious).

My personal want would only be for the broadcast settings (set all the highlighted aliases to use the broadcast setup in a multi-select box -- pretty much just like on the individual alias settings).  But I can see the same desire for bulk adding actions as well.

GTR8000, 2021-05-24:
Have you looked at the Streaming tab in the Playlist Editor, select the stream, then go to the Aliases tab at the bottom of the window? That allows you to bulk add/remove talkgroups from a stream. I could be wrong, but it seems that you're asking for a feature that already exists.

GTR8000, 2021-05-24:
And by the way, you can bulk add/remove talkgroups from being recorded by going to the Record sub-tab on the Aliases page in the Playlist Editor.

ImperatorRuscal, 2021-05-27:
And now I feel a fool.  I was so focused on this as part of the alias setup (color, icon, listen status, broadcast selection) that I completely missed looking deeper into the broadcast tab.
Thanks!

GTR8000, 2021-05-27:
No fools here, we all fumbled through figuring the software out at some point. I believe you can close this ticket out on your own if you feel the issue is resolved.

closed 2021-05-28 by ImperatorRuscal