DSheirer/sdrtrunk #1066: Suggestion: Allow comms history list to scroll from top, not bottom

Brad-git-man, 2021-05-24:
Put a setting in the preferences that allows for the comms history list to scroll down from the top and have the newest comm history added to the top of the list. This would make the comms history list so much more useful than it is now.

liquidmayo, 2021-05-24:
You can sort the events history by timestamp, allowing the newest traffic
to be at the top?

Brad-git-man, 2021-05-24:
@liquidmayo  Where is that setting?

Brad-git-man, 2021-05-25:
@liquidmayo I did notice the if you hit the button "Aged Off" at the bottom of the Events List that it does make recent events scroll form the top down. But it is not residual after program restarts & I'm not even sure if it is just a coincidence of pushing the "Aged Off" button, that it makes top down scrolling.

liquidmayo, 2021-05-25:
Just click the time column header in the events tab

GTR8000, 2021-05-25:
Pretty straightforward...can sort on any of the columns by clicking the header.

[image]
[image]