DSheirer/sdrtrunk #1067: Can't hear conventional (simplex) P25 digital channels

teddy9264, 2021-05-25:
I've got SDRTrunk set up to listen to the Regional Mobile Radio (Victoria, Australia), Phase 2 trunked network, and works great. Now I want to listen to the simplex channels.

I know the frequencies and it shows up on the waterfall when someone is transmitting but regardless of if I choose _P25 phases I_ or _P25 phase II individual_ for the channel, it doesn't make any noise, but I see it on the waterfall.

Any ideas of how to listen to digital simplex channels in SDRTrunk?

infecticide, 2021-06-09:
Are you sure these calls are part of the same P25 network you're monitoring via the Control Channel?
Are you sure it's digital radio?   
Have you tried listening to these calls on SDRSharp to see if they are plain analog radio?

teddy9264, 2021-06-09:
I'm sure they are part to the same network, I know they are digital because I've listened to them on my HT and SDR Sharp and it isn't analog.

ksm-dg, 2024-12-01:
@nic-hanna did you ever get this sorted out? I'm having the same problkem

DSheirer, 2024-12-01:
I did some testing and found that the sdrtrunk P25 decoder has trouble consistently decoding these short bursty transmissions.

I'm currently working on a new DQPSK decoder for DMR that I'll also integrate with P25 next.  This should make it work more reliably for these simplex channels.

ksm-dg, 2024-12-01:
Appreciate the response, thanks!