DSheirer/sdrtrunk #1069: Suggestion: Shorten bottom padding on SDRTrunk

Brad-git-man, 2021-06-13:
There is no reason to waste a 60px high space, all the way across the bottom of SDRTrunk. Ten or fifteen pixels of bottom padding would be plenty. I have been a web designer for over 10 yrs & screen space is a valuable commodity, not to be wasted.

whelgeson, 2021-09-18:
@Brad-git-man Can you post a screenshot for reference?

Brad-git-man, 2021-09-19:
> Can you post a screenshot for reference?

I figured out a few months ago, that all you have to do is take the bottom of the Events lists w/ your mouse and pull it closer to the bottom of the screen.  Like I said, I don't like to waste screen real state. And that the 60px default padding at bottom of SDRT is too much.