DSheirer/sdrtrunk #1070: Suggestion: Add ability to add more than 2 playback channels.

Nokoa, 2021-07-02:
I would love to have the ability to have more than 2 playback channels even if they talk over each other.

Brad-git-man, 2021-07-02:
It sounds like you need more tuners. I have 4 sites running on 3 tuners and of course in just one instance of SDRTrunk. The program supports as many turners as you can afford and as many sites in your area that they can receive.

Ctpolicelive, 2021-07-02:
Na. More tuners will USB fail. Anything over 3 tuners will kick a USB error and fail.

k8tek, 2021-07-10:
> Na. More tuners will USB fail. Anything over 3 tuners will kick a USB error and fail.

Really, my system runs 5. You probably need to give JRE more memory

Brad-git-man, 2021-07-10:
Yeah, I Run 3 tuners & increased the ram to the JVM to 4 gigs, on an 8 core 32 gig system. Below I have posted the link to the RR forum that tells how to increase the ram to the SDRTrunk JVM (Java Virtual Machine) for anyone who wants it.

https://forums.radioreference.com/threads/losing-audio.422394/#post-3470490