DSheirer/sdrtrunk #1073: How do you save column positions & sizes in SDRTrunk?

Brad-git-man, 2021-07-12:
Can anyone say how to save your column positions & sizes that you set in SDRTrunk? (If that is even possible?)  It is very annoying to have to reset your column positions & sizes every time you restart SDRTrunk.

Brad-git-man, 2021-08-04:
The OP here just updating my question. Does anyone know yet how to save your column positions & sizes of SDRTrunk so that they persist through restarts of the program?

GTR8000, 2021-08-05:
No issues here with the column widths being saved on a Windows install. Values are stored in the registry under Computer\HKEY_CURRENT_USER\Software\JavaSoft\Prefs\io\github\dsheirer\preference\swing

Brad-git-man, 2021-08-05:
@GTR8000  I just looked & my registry key ends at "io" with no other sub keys and of course with no user prefs listed. Do you have any idea what could be wrong or how I could remedy it. On another point, the folder I'm running SDRTrunk in is about 4 folders deep in sub folders. Do you think that might be causing it to not be able to save user prefs? Are you running your SDRtrunk in a folder on the root of C: ?

GTR8000, 2021-08-05:
What release of SDRTrunk are you running? What version of Windows?

I run it out of C:\SDRTrunk\, with the user files in the default C:\Users\GTR8000\SDRTrunk\ directory. No issues with it on three separate computers.

Brad-git-man, 2021-08-05:
@GTR8000  I just got it off of here about 3 months ago I think it is the current ver. I am running Win10 LTSC which is 1809.

Brad-git-man, 2021-08-06:
@GTR8000  Even though it still doesn't keep my column settings. I was running it from a drive other then C:. So I moved it's folder over to the root of C: and ran from it there. But it still doesn't create any Registry sub keys deeper the "io" and there is nothing in the Users\*MY User Account*\ SDRTrunk folder. What do you think I should try next?