DSheirer/sdrtrunk #1076: Unable to start SDRTrunk

LunaEros, 2021-08-09:
Don't know if I'm missing something  but it doesn't seem to want to start.  It says something about no device and no devicelistpointer initialized.  I'm on Windows 7 and have an RTL-SDR v3 that works with SDR Sharp, DSD+ and Unitrunker.

This is what the batch file runs....

12:44:43.856 INFO  i.g.d.icon.IconModel - loading icons file [C:\Users\Stormfront\SDRTrunk\settings\icons.xml]  [11MB/256MB 4%]
12:44:43.861 INFO  i.g.d.icon.IconModel - Icons file not found at [C:\Users\Stormfront\SDRTrunk\settings\icons.xml]  [11MB/256MB 4%]
12:44:44.102 INFO  i.g.d.log.ApplicationLog - Application Log File: C:\Users\Stormfront\SDRTrunk\logs\sdrtrunk_app.log  [11MB/256MB 4%]
12:44:44.150 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.5.0-alpha6[5MB/256MB 2%]
12:44:44.151 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 6.3  [5MB/256MB 2%]
12:44:44.151 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2020-10-26T04:29:57.816-0400  [5MB/256MB 2%]
12:44:44.152 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 15 (BellSoft 15+36  [5MB/256MB 2%]
12:44:44.152 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 4.4.0-193-generic  [5MB/256MB 2%]
12:44:44.152 INFO  i.g.d.log.ApplicationLog -   [5MB/256MB 2%]
12:44:44.152 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [5MB/256MB 2%]
12:44:44.152 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [5MB/256MB 2%]
12:44:44.153 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [5MB/256MB 2%]
12:44:44.153 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [5MB/256MB 2%]
12:44:44.153 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [5MB/256MB 2%]
12:44:44.154 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 7[5MB/256MB 2%]
12:44:44.155 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [5MB/256MB 2%]
12:44:44.155 INFO  i.g.d.log.ApplicationLog - Host OS Version:       6.1  [5MB/256MB 2%]
12:44:44.156 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4  [5MB/256MB 2%]
12:44:44.157 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  3 GB  [5MB/256MB 2%]
12:44:44.157 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [5MB/256MB 2%]
12:44:44.158 INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\Stormfront\SDRTrunk  [5MB/256MB 2%]
12:44:44.158 INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\Stormfront\SDRTrunk\logs  [5MB/256MB 2%]
12:44:44.159 INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\Stormfront\SDRTrunk\event_logs  [5MB/256MB 2%]
12:44:44.160 INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\Stormfront\SDRTrunk\playlist  [5MB/256MB 2%]
12:44:44.161 INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\Stormfront\SDRTrunk\recordings  [5MB/256MB 2%]
12:44:44.162 INFO  i.g.d.util.ThreadPool - Application thread pool created with[4] threads  [5MB/256MB 2%]
12:44:44.163 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: C:\Users\Stormfront\SDRTrunk  [5MB/256MB 2%]
12:44:44.164 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\Stormfront\SDRTrunk\SDRTrunk.properties]  [5MB/256MB 2%]
12:44:44.165 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [C:\Users\Stormfront\SDRTrunk\SDRTrunk.properties]  [5MB/256MB 2%]
12:44:44.177 INFO  i.g.d.s.SettingsManager - SettingsManager - settings does not exist [C:\Users\Stormfront\SDRTrunk\settings\settings.xml]  [6MB/256MB 2%]
12:44:44.179 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [6MB/256MB 2%]
12:44:44.270 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478  [8MB/256MB 3%]
12:44:44.271 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312  [8MB/256MB 3%]
12:44:44.325 ERROR i.g.d.s.t.TunerManager - unable to get device list from libusb [-4 / LIBUSB_ERROR_NO_DEVICE]  [8MB/256MB 3%]
Exception in thread "main" java.lang.IllegalStateException: deviceListPointer is not initialized
        at org.usb4java.LibUsb.freeDeviceList(Native Method)
        at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:188)
        at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:79)
        at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:46)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:167)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:624)

Ctpolicelive, 2021-08-09:
Stormfront? lol

ERROR i.g.d.s.t.TunerManager - unable to get device list from libusb [-4 / LIBUSB_ERROR_NO_DEVICE] [8MB/256MB 3%]

Anyway, looks like it's not finding your USB. Did you try Zadig? 
Win 7 is sketchy. I would try with win10 or a linux OS.. or Like Linux Mint or something if you can.

LunaEros, 2021-08-09:
Yeah. I used Zadig to get it to work with SDR Sharp. 
I don't have any other computer to try it on though.

And my email was chosen for an amateur drag car I had years before that website used it.

rapbando, 2022-09-21:
[WINDOWS 7 FIX](https://github.com/usb4java/usb4java/releases/download/usb4java-1.2.0/usb4java-1.2.0.zip)

closed 2022-11-26 by DSheirer