DSheirer/sdrtrunk #1087: Use Specific SDRs

BT-Justice, 2021-10-03:
Is there a way to tell sdrtrunk to only use specific SDRs when opened and ignore all other SDRs hooked up to a computer?  I use 2 RTL-SDR.com V3 SDRs to monitor a trunked system.  These are actually really good SDRs.  I want to use a couple of other ones in other software for other needs like dump1090 and shortwave listening.  I know I can open other software and use the other SDRs which makes them unavailable in sdrtrunk, but I am looking for a more permanent way to tell sdrtrunk specifically which two SDRs I want it to use.

kkomarica, 2022-01-03:
I also have a similar problem and I would also like to use only a certain SDR with SDRTrunk. 
That would be a great feature to have!

hans109h, 2022-01-03:
Duplicate of known feature enhancement: https://github.com/DSheirer/sdrtrunk/issues/119

DSheirer, 2022-04-24:
This feature was added and will be in the next (0.5.0 beta 3) release.

closed 2022-04-24 by DSheirer