DSheirer/sdrtrunk #1111: Null Pointer Exception

LordDonkey, 2021-11-23:
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException: Cannot read field "modelIndex" because "this.viewToModel[index]" is null
        at java.desktop/javax.swing.DefaultRowSorter.convertRowIndexToModel(DefaultRowSorter.java:508)
        at java.desktop/javax.swing.JTable.convertRowIndexToModel(JTable.java:2686)
        at java.desktop/javax.swing.JTable.getValueAt(JTable.java:2763)
        at java.desktop/javax.swing.JTable.prepareRenderer(JTable.java:5780)
        at java.desktop/javax.swing.plaf.basic.BasicTableUI.paintCell(BasicTableUI.java:2210)
        at java.desktop/javax.swing.plaf.basic.BasicTableUI.paintCells(BasicTableUI.java:2112)
        at java.desktop/javax.swing.plaf.basic.BasicTableUI.paint(BasicTableUI.java:1908)
        at java.desktop/javax.swing.plaf.ComponentUI.update(ComponentUI.java:161)
        at java.desktop/javax.swing.JComponent.paintComponent(JComponent.java:842)
        at java.desktop/javax.swing.JComponent.paint(JComponent.java:1119)
        at java.desktop/javax.swing.JComponent.paintToOffscreen(JComponent.java:5311)
        at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBufferedImpl(RepaintManager.java:1657)
        at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1632)
        at java.desktop/javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1570)
        at java.desktop/javax.swing.RepaintManager.paint(RepaintManager.java:1337)
        at java.desktop/javax.swing.JComponent._paintImmediately(JComponent.java:5259)
        at java.desktop/javax.swing.JComponent.paintImmediately(JComponent.java:5069)
        at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:879)
        at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:862)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
        at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:862)
        at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:835)
        at java.desktop/javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:784)
        at java.desktop/javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1898)
        at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:771)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:741)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

smyers119, 2022-02-19:
@LordDonkey is this still present on latest alpha?

LordDonkey, 2022-02-20:
I haven't seen the NPE since, I'm running the Jan 2 commit.

DSheirer, 2023-07-14:
Closing.  Likely resolved earlier, or under PR #1601.

closed 2023-07-14 by DSheirer