DSheirer/sdrtrunk #1117: Bindings.size

sharpentheaxe, 2021-11-28:
Hi,

I am trying to compile the software and I am getting an error when I try to compile the software.  I have fought through many dependency errors and this is my last error.  I have JDK17.0.1 and gradle 7.3.  This is on a window x64 machine.   In the Alias.java file, line 90 I am getting a method size in the is not applicable for the arguments, see message below.   This has been done before, am I missing a library or a setting?

The method size() in the type Map<String,Object> is not applicable for the arguments (new FilteredList<>(mAliasIDs, (<no type> aliasID) -> {
  return ((aliasID != null) && (! aliasID.isAudioIdentifier()));
}))

DSheirer, 2021-11-28:
The Bindings class is part of the JavaFX runtime.

Does the Java 17 JDK that you installed include the JavaFX libraries?  Bellsoft and Azul provide JDKs that include the JavaFX runtimes.

sharpentheaxe, 2021-11-29:
Bellsoft JDK worked very well.   I thought running through the Apache JDK would have all the javafx libraries, apparently not.

 

Very nice software, this looks like it was a lot of work.   I hope to contribute someday.

 

Thank you for your help!

 

Brian

closed 2026-11-06 by DSheirer