DSheirer/sdrtrunk #1123: Audio segment detected with NO TO Identifiers

simplio, 2021-12-03:
Some audio skips and garbled audio with the error 4 times in the console log of SDRTrunk. It clears itself and intermittent. I tried to google this but not finding a straight to the point remedy or cause. What I have found was mentions of excessive teardowns. I do not know what to do. I seem to come up at a dead end as Google searching and radioreference searching brings up nothing. Please help
[image]

I tried to increase Java stack heap to 1gb and 2gb. It was futile. Errors persist at a intermittent rate.

DSheirer, 2021-12-03:
This can happen when there is poor decode quality.  The decoder was able to decode and produce audio, but it wasn't able to capture any identifiers related to the audio segment.  

This log entry is informational to let you know that the recording manager won't be able to include any identifiers in the audio recording filename for that audio segment.

efirstalert, 2021-12-26:
I have been seeing a large number of the same reports.  From the report above that it's related to poor decodes, I decided to play with the gain settings on my SDRs.  I changed them from 323 (which is apparently the default, because I never set it before) to "Automatic" and initial tests are looking tremendously better.

efirstalert, 2021-12-26:
Follow up.  I was getting the error message at least once every couple of minutes.  Since I changed the gain to automatic it's been running a couple hours with ZERO errors, and my audio quality is greatly improved.