DSheirer/sdrtrunk #1124: Feature Request - Volume Control based on Talkgroup

tbitson, 2021-12-11:
First off - Thank you for an awesome app. I use this daily. 

The local fire dispatch uses tones to signal events. These are very loud. The Local PD dispatch talk very quietly. I can never found a volume level that works - therefore it would be a nice feature if the volume could be set in each talkgroup independently.

Thanks again for your effort!

liquidmayo, 2021-12-13:
Just commenting for no other reason but to "second" this.   Have this exact same issue pretty much to the letter.

boteman, 2021-12-14:
I have had good success with VSThost running 2 plug-ins to shape the audio and act as an ALC since P25 audio seems to vary by 25-30 dB from radio to radio. I use TDR Nova for a parametric equalizer and TriLeveler2 to even out the shouters and whisperers.

If SDRtrunk had a drop-and-insert function this audio processing could be inserted between the decoded output and the stream sent to a host, but would work fine for local listening using a virtual audio cable. This would seem to be a better alternative than re-inventing the wheel inside SDRtrunk.