DSheirer/sdrtrunk #1146: Release a new stable version

blshkv, 2022-02-22:
Hello,

could you please cut a stable release at some point? The last version was released back in 2020 but there were many important fixes since then.

k8tek, 2022-03-05:
[sdr-trunk-0.5.0-alpha7.zip](https://github.com/DSheirer/sdrtrunk/files/8189417/sdr-trunk-0.5.0-alpha7.zip)
You can always compile it yourself.

blshkv, 2022-03-06:
even this alpha version is old too, released back in Oct 26, 2020.

I'm asking to release a new version for all linux distros to pick it up. 
P.S. I'm a Pentoo developer.

k8tek, 2022-03-06:
You don’t compile code as a developer?  Everything is there for you to do it yourself.

blshkv, 2022-03-06:
Gentoo is source based distribution. So not only developers but even users compile it. 

But this is not the point. Our users want a stable release, not a random snapshot

dechilders, 2022-03-06:
He already said on the google group that a new release is coming shortly.

Brad-git-man, 2022-03-07:
> He already said on the google group that a new release is coming shortly.

Can anyone here say how to upgrade when he does release it?  I am guessing that it has no upgrade ability and you delete your current install. I truly hope you tell me, I'm wrong.

k8tek, 2022-03-07:
I usually back up the directory I installed it in and then I just copy the new files.

blshkv, 2022-07-28:
and by shortly he means shortly, right?

closed 2022-11-26 by DSheirer