DSheirer/sdrtrunk #1147: Incorrect Talkgroup

ispawnidie, 2022-02-26:
I'm using sdrtrunk alpha 6 on a windows pc. Everything is working great except I'm getting some dispatch tone outs coming up as the incorrect talkgroup. It's only on the initial tone out, all other traffic is fine. It also seems to be one particular dispatch center, I don't see any issues with the other dispatch centers on the system. I can confirm that they are broadcasting on the correct talkgroup using other software (op25) and a Unication G5 pager.

System being monitored [https://www.radioreference.com/apps/db/?sid=7643](https://www.radioreference.com/apps/db/?sid=7643)

boteman, 2022-02-26:
Info: I have witnessed unexpected behavior similar to this on other trunked systems (on Uniden scanners and UniTrunker), specifically related to dispatches. It's possible that the tones are sent via a multi-select group, then the voice broadcast is sent over the "normal" talkgroup. 

Is the incorrect talkgroup consistently the same or some random number?

Of course, on my local P25 Phase 2 system I often hear ambulance reports to the hospital on the local speaker with no talkgroup label whatsoever, so it's possible that SDRtrunk is not decoding things properly.

ispawnidie, 2022-02-26:
The talkgroup always starts with 651 and the last two digits seem to be random. It's the whole tone out and dispatch info (Station, address, ect.) all other traffic is fine (unit to unit, dispatch to unit, unit to dispatch).

ispawnidie, 2022-03-03:
I found out that it is a dynamic talkgroup that is part of a multi group. I don't know what the purpose is but there is still the issue that sdrtrunk is only picking it up as the dynamic talkgroup and not the talkgroups that I am streaming.

ispawnidie, 2022-03-21:
This is still an issue with 0.5.0 Beta 1 and Beta 2

ispawnidie, 2023-01-02:
Apparently my issue is related to #1325

thesea-is-foamy, 2023-01-30:
This issue is present on harris systems in 0.5.0 final and 0.5.1 beta 1.  Traffic ends up being recorded with only the patch's talkgroup id, not the _patched_ talkgroups.  This creates a problem where you don't know what talkgroups are actually talking.

[image]

The behavior looks correct on moto systems though.

[image]

Seagravebuff60, 2023-02-16:
I seem to be having the same issue here with SDR Trunk. This is specificly true for haris systems only. When a soft patch is started, the conversation happens on the 65xxx talkgroup, and with SDR Trunk at least you can not tell the orginal talkgroups that are patched. Unlike with motorola systems when a soft patch is started you can correctly folow the patch. 

The 65xxx talkgroup becomes especialy complicated when trying to narrow down exact soft patches for new systems and for Broadcasting systems to 3rd party streamers like broadcastify calls.