DSheirer/sdrtrunk #1149: Latest update broken (03/04/2022)

k8tek, 2022-03-04:
I tried compiling the update that was released today and it has several issues.

`> Configure project :
Skipping OS Image - Linux ARM 64-bit JDK was not found at /home/denny/java_jdks/linux-arm64/jdk-17.0.2-full
Skipping OS Image - Linux x86 64-bit JDK was not found at /home/denny/java_jdks/linux-x64/jdk-17.0.2-full
Skipping OS Image - OSX x86 64-bit JDK was not found at /home/denny/java_jdks/osx-x64/jdk-17.0.2-full.jdk
Skipping OS Image - Windows x86 64-bit JDK was not found at /home/denny/java_jdks/windows-x64/jdk-17.0.2-full

> Task :compileJava
warning: using incubating module(s): jdk.incubator.vector
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:30: error: package javafx.beans does not exist
import javafx.beans.Observable;
                   ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:31: error: package javafx.beans.property does not exist
import javafx.beans.property.SimpleStringProperty;
                            ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:32: error: package javafx.util does not exist
import javafx.util.Callback;
                  ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:47: error: cannot find symbol
    private SimpleStringProperty mValueProperty = new SimpleStringProperty();
            ^
  symbol:   class SimpleStringProperty
  location: class AliasAction
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:57: error: cannot find symbol
    public SimpleStringProperty valueProperty()
           ^
  symbol:   class SimpleStringProperty
  location: class AliasAction
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:33: error: package javafx.beans does not exist
import javafx.beans.Observable;
                   ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:34: error: package javafx.beans.binding does not exist
import javafx.beans.binding.Bindings;
                           ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:35: error: package javafx.beans.binding does not exist
import javafx.beans.binding.BooleanBinding;
                           ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:36: error: package javafx.beans.property does not exist
import javafx.beans.property.BooleanProperty;
                            ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:37: error: package javafx.beans.property does not exist
import javafx.beans.property.IntegerProperty;
                            ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:38: error: package javafx.beans.property does not exist
import javafx.beans.property.SimpleBooleanProperty;
                            ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:39: error: package javafx.beans.property does not exist
import javafx.beans.property.SimpleIntegerProperty;
                            ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:40: error: package javafx.beans.property does not exist
import javafx.beans.property.SimpleStringProperty;
                            ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:41: error: package javafx.beans.property does not exist
import javafx.beans.property.StringProperty;
                            ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:42: error: package javafx.collections does not exist
import javafx.collections.FXCollections;
                         ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:43: error: package javafx.collections does not exist
import javafx.collections.ObservableList;
                         ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:44: error: package javafx.collections.transformation does not exist
import javafx.collections.transformation.FilteredList;
                                        ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:45: error: package javafx.util does not exist
import javafx.util.Callback;
                  ^
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:93: error: cannot find symbol
    public static Callback<AliasAction,Observable[]> extractor()
[... 334 lines cut ...]
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\audio\AudioSegment.java:71: error: cannot find symbol
    private ObservableSet<BroadcastChannel> mBroadcastChannels = FXCollections.observableSet(new HashSet<>());
            ^
  symbol:   class ObservableSet
  location: class AudioSegment
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\audio\AudioSegment.java:136: error: cannot find symbol
    public BooleanProperty completeProperty()
           ^
  symbol:   class BooleanProperty
  location: class AudioSegment
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\audio\AudioSegment.java:145: error: cannot find symbol
    public BooleanProperty duplicateProperty()
           ^
  symbol:   class BooleanProperty
  location: class AudioSegment
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\audio\AudioSegment.java:158: error: cannot find symbol
    public ObservableSet<BroadcastChannel> broadcastChannelsProperty()
           ^
  symbol:   class ObservableSet
  location: class AudioSegment
C:\sdrtrunk-master\src\main\java\io\github\dsheirer\audio\AudioSegment.java:186: error: cannot find symbol
    public BooleanProperty recordAudioProperty()
           ^
  symbol:   class BooleanProperty
  location: class AudioSegment`

hans109h, 2022-03-04:
I had the same issue and Denny reminded me I need Bellsoft Liberica 'Full' JDK or the Azul 'Full' JDK that includes the JavaFX libraries.

Since I already had Bellsoft it reminded me that on my system I needed to set JAVA_HOME to point to the right JDK.

k8tek, 2022-03-05:
Nevermind, I deleted gradle and it compiled.

closed 2022-03-05 by k8tek