DSheirer/sdrtrunk #1156: JMBEAudioLibrary compiled by more recent version of Java Runtime

entertwined, 2022-03-07:
It appears that the version of JMBE created by JMBE Creator is not compatible with the version of Java bundled with the Windows download for 0.5.0 Alpha 6? Or something along those lines, to be honest Java is a complete mystery to me. When trying to create the JMBE library I get the following error, and audio doesn't work.

`Exception in thread "JavaFX Application Thread" java.lang.UnsupportedClassVersionError: jmbe/JMBEAudioLibrary has been compiled by a more recent version of the Java Runtime (class file version 61.0), this version of the Java Runtime only recognizes class file versions up to 59.0`

and then a long traceback.

hans109h, 2022-03-07:
JavaFX is not included in all JDK's.  Denny suggests either the Bellsoft Liberica 'Full' JDK or the Azul 'Full' JDK as both of these include JavaFX libraries.

entertwined, 2022-03-07:
I'm sorry, I must be misunderstanding something. I don't have any Java installed. According to the release notes, 

> "Starting with Version 0.5.0 Alpha 2, you no longer have to install Java and compile the JMBE library. sdrtrunk now has integrated support for creating the JMBE library."

There is a java.exe bundled with the Windows release. I did try installing the "Azul Zulu JDK FX" and running the program with that version of Java, but then the application wouldn't even launch (got some kind of GUI related error I think).

dalarso, 2022-03-07:
I have this same issue. I reinstalled jdk-17.0.2 and I cannot get passed this error.

hans109h, 2022-03-07:
Sorry, I don't have any experience with Windows, but can you both please clarify your source of sdrtrunk (ie. are you using an already compiled release or are you compiling yourself)?

entertwined, 2022-03-08:
I was attempting to use the compiled Windows release included in the "Assets" section fromVersion 0.5.0 Alpha 6: [sdr-trunk-windows-x86_64-v0.5.0-alpha6.zip](https://github.com/DSheirer/sdrtrunk/releases/download/v0.5.0-alpha.6/sdr-trunk-windows-x86_64-v0.5.0-alpha6.zip) As far as I can tell, it includes a copy of the Java runtime, and a batch script to launch the application that overrides your JAVA_HOME to use the bundled Java by default. Somehow it appears this bundled version of Java is not compatible with the object created by the JBME creator component.

I decided to just try building sdrtrunk per the instructions, installed gradle and the Azul17.0.2 FX JDK, now everything works fine.

dalarso, 2022-03-08:
I was able to finally get mine running by uninstalling jdk-17.0.2 and installing OpenJDK 16 instead. I also directly downloaded JMBE Library version 1.0.7b (instead of 1.0.8) from https://github.com/DSheirer/jmbe/releases and ran the creator.bat script. Pointed str-trunk to the generated JMBE file and its now working for me.

hans109h, 2022-03-08:
I'm glad you figured it out.  The newest JMBE library was built against a later java JDK than was used to build the asset you were using, so it was doomed to fail.  For others that experience this they will either need to build the JMBE library outside of sdrtrunk, compile the latest sdrtrunk from source code or wait until a new binary is released.

DSheirer, 2022-03-08:
This should be resolved.  I re-released the JMBE Creator as version 1.0.9 and it now uses JDK 11 to build the library.  This should be compatible with sdrtrunk back through version 0.5.0 alpha 2.

closed 2022-03-08 by DSheirer