DSheirer/sdrtrunk #1162: RX-mode hackRF One firmware 2021.03.1

ah1102, 2022-03-12:
hackRF One on the latest firmware 2021.03.1 stopped disabling receive mode when exiting sdrtrunk 0.5.0 alpha 6. This is indicated by the LED indicator. The sdrtrunk program does not reconnect to hackrf one and you need to reset the device or turn it off / on.

scooterusa, 2022-03-14:
Adding to this.. It isn't that SDRTrunk is disabling receive mode when exiting. When exiting SDRTrunk it just doesn't send the "stop" command (or equivalent) which results in the hackRF one still being in RX mode. This is why you end up having to stop the RX mode when exiting SDRTrunk is by resetting the hackRF one.

SDRTrunk can send the "stop" command to the hackRF one if you go into playlist editor, then to channels, select the active channel and then press the stop button. At this point you can exit SDRTrunk, restart SDRTrunk and resume scanning without issue.

Perhaps it maybe possible to add the same (or similar) "stop" command to execute when exiting SDRTrunk.

DSheirer, 2022-03-15:
sdrtrunk does attemptto shutdown each of the tuners before the application exits. I'll have to look at the updates for this firmware to see if something has changed.

alphafox02, 2022-03-25:
I just observed the same thing w/ 2021.03.1 and loaded on a HackRF and associated libhackrf source version in 20.04. When I first start 0.5.0 Beta2 the HackRF is immediately in use with a working fft/waterfall. If I exit the application either by x'ing out and clicking exit, the RF light remains on. 

If I run HackRF_info, the Rx light will go out and the HackRF is usable again. If I exit SDRTrunk and then try to start it again without unplugging the HackRF and running a utility against it, it's unusable to include not starting within SDRTrunk.

scooterusa, 2022-03-25:
The issue that we are having is that SDRTrunk is not sending the stop command for RX functions. Claiming that it is "unusable" or "disabled" is not accurate nor true. That is improper terminology for what is going on.

Adding a little more to the matter (in case anyone else has the same issue, and noting a new change).. 

Launching SDRTrunk and starting a playlist entry we can see that the RX is enabled, we can see that it doesn't stop when exiting SDRTrunk. 

Under the previous Beta we could stop the HackRF One with the playlist editor, however under this beta it no longer works. SDRTrunk will say "idle" for the control channel and the HackRF One remains in RX mode. At this point two things can happen:

1. When pressing play without exiting SDRTrunk the application will resume picking up the control channel because the HackRF One is still in RX Mode.
2. Exiting and restarting SDRTrunk will result in the control channel not being detected because the HackRF One never received the RX stop command prior to exiting SDRTrunk

Using external commands/tools like hackRF_info (or equivalent) that send the stop command (or resetting the unit itself) will cause the unit to stop the RX mode. Starting SDRTrunk when the RX was stopped will (of course) result in the application detecting the control channel. Of course the same issue of us having to manually reset the device or use another means to stop the RX when we have to restart SDRTrunk is still applicable. 

Again all this means is that SDRTrunk is simply just not sending the stop command. Our HackRF One units are not disabled nor are they unusable... 

I am sure it is being looked into and a resolution will be deployed as soon as it can be. For now we just need to reset them or manually issue the stop command (whether command line or etc) before restarting SDRTrunk.

DSheirer, 2022-04-17:
Issue is addressed under #1190

closed 2022-04-17 by DSheirer