DSheirer/sdrtrunk #1181: Apple Silicon Mac M1 MacBook PRO beta problem

DanielAbsi, 2022-03-19:
I try to install SDR Trunk on my macbook pro M1. But isn't working 
**This is what I have:**
1. SDR: RTL-SDR 
2. java 17.0.2
3. JDK 17.0.2+9 ARM 64 bit for macOS

When running sdr-trunk this is that I get: 
``` 
WARNING: Using incubator modules: jdk.incubator.vector
WARNING: package com.sun.java.swing.plaf.windows not in java.desktop
14:20:22.539 INFO  i.g.d.icon.IconModel - loading icons file [/mac-users/mac-user/SDRTrunk/settings/icons.xml]  [10MB/256MB 3%]
14:20:22.542 INFO  i.g.d.icon.IconModel - Icons file not found at [/mac-users/mac-user/SDRTrunk/settings/icons.xml]  [10MB/256MB 4%]
14:20:22.773 INFO  i.g.d.log.ApplicationLog - Application Log File: /mac-users/mac-user/SDRTrunk/logs/sdrtrunk_app.log  [17MB/256MB 6%]
14:20:22.783 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.5.0-beta1  [18MB/256MB 7%]
14:20:22.783 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 7.4  [18MB/256MB 7%]
14:20:22.783 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2022-03-18T11:57:38.285-0400  [18MB/256MB 7%]
14:20:22.783 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 17.0.2 (BellSoft 17.0.2+9-LTS  [18MB/256MB 7%]
14:20:22.783 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 5.13.0-35-generic  [18MB/256MB 7%]
14:20:22.783 INFO  i.g.d.log.ApplicationLog -   [18MB/256MB 7%]
14:20:22.783 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [18MB/256MB 7%]
14:20:22.784 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [18MB/256MB 7%]
14:20:22.784 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [18MB/256MB 7%]
14:20:22.784 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [18MB/256MB 7%]
14:20:22.784 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [18MB/256MB 7%]
14:20:22.784 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Mac OS X  [18MB/256MB 7%]
14:20:22.784 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          aarch64  [18MB/256MB 7%]
14:20:22.784 INFO  i.g.d.log.ApplicationLog - Host OS Version:       12.3  [18MB/256MB 7%]
14:20:22.784 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        8  [18MB/256MB 7%]
14:20:22.784 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  4 GB  [18MB/256MB 7%]
14:20:22.785 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [18MB/256MB 7%]
14:20:22.785 INFO  i.g.d.log.ApplicationLog -  Application Root: /mac-users/mac-user/SDRTrunk  [18MB/256MB 7%]
14:20:22.785 INFO  i.g.d.log.ApplicationLog -  Application Log:  /mac-users/mac-user/SDRTrunk/logs  [18MB/256MB 7%]
14:20:22.785 INFO  i.g.d.log.ApplicationLog -  Event Log:        /mac-users/mac-user/SDRTrunk/event_logs  [18MB/256MB 7%]
14:20:22.785 INFO  i.g.d.log.ApplicationLog -  Playlist:         /mac-users/mac-user/SDRTrunk/playlist  [18MB/256MB 7%]
14:20:22.785 INFO  i.g.d.log.ApplicationLog -  Recordings:       /mac-users/mac-user/SDRTrunk/recordings  [18MB/256MB 7%]
14:20:22.831 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [22MB/256MB 8%]
14:20:22.831 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /mac-users/mac-user/SDRTrunk  [22MB/256MB 8%]
14:20:22.831 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/mac-users/mac-user/SDRTrunk/SDRTrunk.properties]  [22MB/256MB 8%]
14:20:22.831 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/mac-users/mac-user/SDRTrunk/SDRTrunk.properties]  [22MB/256MB 8%]
14:20:22.841 INFO  i.g.d.s.SettingsManager - SettingsManager - settings does not exist [/mac-users/mac-user/SDRTrunk/settings/settings.xml]  [5MB/256MB 2%]
14:20:22.842 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [6MB/256MB 2%]
Exception in thread "main" java.lang.UnsatisfiedLinkError: Can't load library: /var/folders/g3/154q1f492znbdz3dw4x6dlyh0000gn/T/usb4java2733755084416904189.tmp/libusb4java.dylib
	at java.base/java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.base/java.lang.Runtime.load0(Unknown Source)
	at java.base/java.lang.System.load(Unknown Source)
	at org.usb4java.Loader.load(Loader.java:323)
	at org.usb4java.LibUsb.<clinit>(LibUsb.java:690)
	at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:110)
	at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:79)
	at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:46)
	at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:161)
	at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:645)
```
On one of the last line I get this message: 
```
Exception in thread "main" java.lang.UnsatisfiedLinkError: Can't load library: /var/folders/g3/154q1f492znbdz3dw4x6dlyh0000gn/T/usb4java2733755084416904189.tmp/libusb4java.dylib
```
How could I fix the error of sdr-trunk-osx-aarch64-v0.5.0-beta1. 
Thanks

DSheirer, 2022-03-22:
It looks like OSX's sandboxing may be preventing sdrtrunk from using this library.  I may have to embed the library into the sdrtrunk main library so that it's accessible.

nuclearace, 2022-03-26:
> I may have to embed the library into the sdrtrunk application bundle so that it's accessible.

Is there an easy way to test this out for myself? It's been many moon since I've messed around with a gradle built project, so not sure where to mess around with trying to fix this issue.

mayersj1, 2022-03-28:
Seeing the same issue here. Any ETA for a solution? Thanks!

grantbarrett, 2022-04-10:
I can confirm : I am having the same problem. I am also unable to build on macOS 12.3.1 aarch64 using the latest code, but I will report that under a separate issue.

philippjbauer, 2022-04-17:
Same issue here. +1

rabarar, 2022-05-09:
Same issue here. +2

rscullin, 2022-05-10:
This isn't a solution, but may be a workaround until the issue is resolved.

### Workaround

- Make sure you have a M1-native `brew` installed
- Run `brew install libusb`
- Run `sudo mkdir -p /opt/local/lib/ && sudo ln -s /opt/homebrew/lib/libusb-1.0.0.dylib /opt/local/lib/libusb-1.0.0.dylib`
- Launch SDRTrunk via `./gradlew run`

### Details

SDRTrunk in it's errored state prints:

```
Exception in thread "main" java.lang.UnsatisfiedLinkError: Can't load library: /var/folders/n5/h5c489z15q70tnn8c44thp040000gn/T/usb4java13358785594645869030.tmp/libusb4java.dylib
        at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2393)
        at java.base/java.lang.Runtime.load0(Runtime.java:755)
        at java.base/java.lang.System.load(System.java:1953)
        at org.usb4java.Loader.load(Loader.java:323)
        at org.usb4java.LibUsb.<clinit>(LibUsb.java:690)
        at io.github.dsheirer.source.tuner.manager.TunerManager.start(TunerManager.java:99)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:177)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:714)
```

`libusb4java.dylib` exists at that location, but it tries referencing `libusb-1.0.0.dylib` at a (hardcoded?) path that doesn't exist:

```
$ otool -L  /var/folders/n5/h5c489z15q70tnn8c44thp040000gn/T/usb4java13358785594645869030.tmp/libusb4java.dylib
/var/folders/n5/h5c489z15q70tnn8c44thp040000gn/T/usb4java13358785594645869030.tmp/libusb4java.dylib:
	libusb4java.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
```

By creating the directory and linking `libusb-1.0.0.dylib` installed by `brew` to the expected location, SDRTrunk launches and runs fine as a native M1 app.

I'm unsure if this is a sandboxing issue or an issue with the build of `libusb4java.dylib`. Either way, this should help until it's properly resolved.

rabarar, 2022-05-10:
This is a workaround indeed. I can’t get the code to compile (not sure why..) but running the Asset release after linking works fine!

DSheirer, 2022-05-15:
@rscullin It seems that the original issue is that the libusb library was not installed on the machine, or it was not installed in a location that sdrtrunk was expecting to find it?

Does your workaround of installing libusb using homebrew and creating a symbolic link allow you to run the release version of sdrtrunk?

rabarar, 2022-05-15:
I don’t believe that is the issue, although creating a symbolic link will resolve it. The issue is you can’t load a dynamic library from the location where it is installed because of the sandboxing the operating system is going to prevent malicious loading.

cfc62, 2022-05-15:
@Denny same error as always, in looking at the changes it looks like you're looking to inform about the workaround (which I haven't tried yet), but there's nothing different in the logging that I saw.

denny, 2022-05-15:
I don't understand why I've been tagged in this...?

denny, 2022-05-15:
Oh, I see. Hi all, be more careful with your tags in future maybe? Cheers! :-)

cfc62, 2022-05-15:
Whoops not on Discord!

In any case, I applied the workaround (which is extremely time consuming as it involves installing Xcode and command line tools) and it did not work - same error as before...

cfc62, 2022-05-15:
My Brew install put the library into /usr/local/Cellar/libusb/1.0.26/lib/libusb-1.0.0.dylib but linking that doesn't make a difference. 

Note, I looked at the xattrs for the dylib and there are no attributes attached to it at all. That is, there are no quarantine flags set.

rscullin, 2022-05-15:
> @rscullin It seems that the original issue is that the libusb library was not installed on the machine, or it was not installed in a location that sdrtrunk was expecting to find it?
>
> Does your [...]

The original issue seems to be that `libusb4java.dylib` is looking for `libusb-1.0.0.dylib` in a location that doesn't exist on my machine (or wasn't shipped with the library?), even though I'd previously installed it with `brew`.

I'm not sure how this works on x64 OS X, though, as I assume it it's worked fine on that -- maybe that version bundles `libusb-1.0.0.dylib`, and the aarch64 build doesn't?

Installing `libusb` via `brew` and creating the symlink also makes the release version of SDRTrunk (`sdr-trunk-osx-aarch64-v0.5.0-beta3`) work on my M1.

---

> My Brew install put the library into /usr/local/Cellar/libusb/1.0.26/lib/libusb-1.0.0.dylib but linking that doesn't make a difference.

You might be running the x64 version of `brew`, as M1 versions of `brew` default to installing to `/opt/homebrew`.

cfc62, 2022-05-15:
@rscullin - I just installed brew today; this machine was scratch installed a few weeks ago - it's never had brew on it previously.

m1@m1 / % brew -v
Homebrew 3.4.11
Homebrew/homebrew-core (git revision f17b20a1f67; last commit 2022-05-15)
Homebrew/homebrew-cask (git revision 90f8a65189; last commit 2022-05-15)

However, to your point:

m1@m1 / % brew doctor
Your system is ready to brew.
m1@m1 / % brew config
HOMEBREW_VERSION: 3.4.11
ORIGIN: https://github.com/Homebrew/brew
HEAD: 30e3d8883cde265cbc16ff94b49a3ac14137590e
Last commit: 5 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: f17b20a1f67987c677e22e261aab1c67052a8628
Core tap last commit: 4 hours ago
Core tap branch: master
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_CORE_GIT_REMOTE: https://github.com/Homebrew/homebrew-core
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 2.6.8 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: octa-core 64-bit westmere
Clang: 13.1.6 build 1316
Git: 2.32.0 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.79.1 => /usr/bin/curl
macOS: 12.3.1-x86_64
CLT: 13.4.0.0.1.1651278267
Xcode: 13.3.1
Rosetta 2: true

Crap. I figured the brew install process would figure out the chipset but clearly something went amiss here. Thanks for the pointer...

cfc62, 2022-05-15:
OK - even with a clean/scratch install the checkbox for Rosetta 2 for Terminal was set, which was the problem. The brew install script maybe should have caught that? Oh well. 

Workaround works! 

:^)

Note - I was conservative and installed the entirety of Xcode, which is a heavy lift (more than a dozen gigs, can be 1-2 hours overall even on a fast link); if only command line tools are needed that would be good to know. I may test it out if I have time...

Thanks @rscullin and @DSheirer !

rscullin, 2022-05-16:
@cfc62 

> Workaround works!

:tada:

---

> if only command line tools are needed that would be good to know

I believe [this is the case](https://docs.brew.sh/Installation#macos-requirements) for installing `brew`, which is all that you need to install `libusb`.

`otool` may require the full Xcode install, but fortunately that's not needed for the workaround.

---

> Edited to add - ah shoot, the JMBE Library Creator process for the PII calls fails:

Another workaround! (And hopefully the last...) This probably should be it's own issue, but does loosely fall into getting SDRTrunk to run on the M1.

Download the [JMBE Creator Application Release](https://github.com/DSheirer/jmbe/releases) zip, then in a Terminal with the downloaded zip:

```sh
unzip jmbe-creator-osx-aarch64-v1.0.9.zip
cd creator-osx-aarch64-v1.0.9/bin
xattr -r -d com.apple.quarantine  .
./creator
```
Should get you `jmbe-1.0.9.jar` in the parent directory, which you can then tell SDRTrunk to use.

rscullin, 2022-05-18:
After looking at the libraries that are causing issues, I think I've figured out the issue / how to fix it, so that the `brew` / `libusb` workaround above isn't needed any more.

I believe that @DSheirer 's build of `libusb4java` dynamically links to `libusb`, while normally it should be statically linked. I rebuilt `libusb4java`, and the resulting `jar` works without the symlink workaround on my M1, for both my dev build, and as a drop-in jar with 0.5.0-beta3.

There's a reasonable chance that this isn't the most correct way to make it work, but it is progress.

---
# Building `libusb4java`

Building `libusb4java` on my M1 wasn't straightforward, so here's what I did:

```
export JAVA_HOME=`/usr/libexec/java_home`
git clone https://github.com/usb4java/libusb4java
cd libusb4java
```

Edit `CMakeLists.txt` and add the following after the line `set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/.cmake/Modules/")` :

```
if(APPLE)
   set(JAVA_AWT_LIBRARY "$ENV{JAVA_HOME}/lib/libjawt.dylib")
   set(JAVA_JVM_LIBRARY "$ENV{JAVA_HOME}/lib/server/libjvm. libjawt.dylib")
   set(JAVA_INCLUDE_PATH "$ENV{JAVA_HOME}/include")
   set(JAVA_INCLUDE_PATH2 "$ENV{JAVA_HOME}/include/darwin")
   set(JAVA_AWT_INCLUDE_PATH "$ENV{JAVA_HOME}/include")
endif()
```

```
cd dists/darwin
```

Edit `build` to modify the following line (comment out/remove `--host=$ARCH`):
`    ./configure --disable-shared --enable-static --with-pic --prefix="$ROOT_DIR" #--host=$ARCH`

**Note:** I'm not sure _why_ this makes things "work" -- without removing `--host=$ARCH`, `libusb` builds, but it doesn't see any USB devices. The script still builds both an `x86_64` and an `arm64` version of the `libusb` and `libusb4java`, though. Here be dragons.

```
./build
# Build should happen

cd ../../target/

# Get the full file path
readlink -f libusb4java-darwin.jar
```

# Using `libusb4java-darwin.jar` locally

In SDRTrunk's `build.gradle`, add/edit the following:

``` gradle
//    implementation 'io.github.dsheirer:libusb4java-darwin-aarch64:1.3.0' //usb4java native lib for OSX M1 cpu
    implementation files('/path/from/readline/command/above/libusb4java/target/libusb4java-darwin.jar')
```

and then run with `./gradlew run`.

Alternatively, copy `libusb4java-darwin.jar` from above into the `lib` directory of a SDRTrunk Release, and delete `libusb4java-darwin-aarch64-1.3.0.jar`.

# Testing / Confirmation

To confirm that I'm using the locally built version of `libusb` and not my system version, I've removed the symlink, and SDRTrunk launches and shows the waterfall for my tuner. Additionally, my `brew` installed version is `1.0.26`, while the reported version is `1.0.24`, which matches the version built by the `build` script.

```
$ brew info libusb
libusb: stable 1.0.26 (bottled), HEAD

$ ls /opt/local/
ls: /opt/local/: No such file or directory
```

Condensed logs:
```
21:48:50.295 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.5.0-beta3  [19MB/256MB 7%]
21:48:50.297 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Mac OS X  [20MB/256MB 7%]
21:48:50.297 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          aarch64  [20MB/256MB 7%]
21:48:50.297 INFO  i.g.d.log.ApplicationLog - Host OS Version:       12.3.1  [20MB/256MB 7%]
21:48:50.297 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        8  [20MB/256MB 7%]
21:48:50.560 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.264  [14MB/34MB 41%]
21:48:50.560 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.24.11584  [14MB/34MB 41%]
```

DSheirer, 2022-05-19:
@rscullin Can you please send me that 'libusb4java-darwin.jar file?  I'll replace the osx aarch64 jar with that file that you compiled and we can use that moving forward.

I didn't compile the original one ... someone else did.

rscullin, 2022-05-23:
@DSheirer Sure -- I was able to get a functional version of `libusb4java` building with GitHub Actions, so it should be reproducible for future builds. You can download the artifact from the [last run](https://github.com/rscullin/libusb4java/actions/runs/2369392966)  in my [forked `libusb4java` repo](https://github.com/rscullin/libusb4java/). It's a multi arch (`x86_64` and `aarch64`) `.jar`, but SDRTrunk seems to be ok with both it and `libusb4java-1.3.0-darwin-x86-64.jar` co-existing. 

# libusb tangent
`libusb` seems to default to using the `null` USB driver (`os/null_usb.lo`)  if `configure` doesn't get a `--arch` that it understands, which explains the mysterious issues I had previously. Passing a specific arch that `configure` (`x86_64-apple-darwin` or `arm64-apple-darwin`) recognizes results in a functional `libusb` build.

DSheirer, 2022-05-23:
@rscullin I used the jar from your github action build, removed the x86-64 pieces and then published the jar to my sonotype repository.  I updated the sdrtrunk gradle build to use the new jar.

If you have an opportunity, could you fetch the latest master branch and test to see if this continues to work for you?

rscullin, 2022-05-24:
@DSheirer Latest master works!

DSheirer, 2022-05-24:
Closing this issue.  The issue with JMBE creator running on OSX should be resolved with this PR that was merged to master branch: https://github.com/DSheirer/sdrtrunk/pull/1262

closed 2022-05-24 by DSheirer