DSheirer/sdrtrunk #1188: P25 ISSI Patch Talkgroup Aliasing

DSheirer, 2022-03-22:
P25 (I & II) currently only supports aliasing to the TALKGROUP or RADIOID level.  When a cross-system ISSI patch call is encountered, sdrtrunk attempts to alias the external SYSID radio ID to the local pool of aliased radio ID values, resulting in an alias mismatch.

sdrtrunk already has support for fully qualified RADIOID and TALKGROUP on the identifier side.  Add aliasing support for fully qualified P25 Alias Identifiers that require the user to specify both a SYSID and a RADIOID/TALKGROUP value.

DSheirer, 2022-03-22:
Note: is there a way to allow users to export existing aliases that are not fully qualified, to another alias list as fully qualified aliases?  This would require the user to specify the WACN and SYSID prefix values.

ImagoTrigger, 2022-03-23:
possibly allow loading multiple playlist xml files unique to sysid and/or alias group name

PR #1751 (2023-11-29): #1188 APCO25 (ISSI) Fully Qualified Radio & Fully Qualified Talkgroup Aliasing
Closes #1188 

Adds support for aliasing of P25 fully qualified (wacn + system + id) radio and talkgroup identifiers.

Two new menu options are added in the playlist editor for adding fully qualified radio and talkgroup alias identifiers to an alias.

This should enable correct aliasing of these values in recording, streaming, and playback/display.
closed 2023-11-29 by DSheirer