DSheirer/sdrtrunk #1193: Unable to source channel from preferred tuner

Varean, 2022-03-26:
SDR Trunk starts correctly, I see the waterfall and frequencies, the tuner is found and loaded, but when I try to play the control channel I get an error that a tuner is not available. I've attached my log. 
[image]
[sdrtrunk_app_upload.log](https://github.com/DSheirer/sdrtrunk/files/8354340/sdrtrunk_app_upload.log)

DSheirer, 2022-03-26:
There's something bad going on with either your RTL tuner or the USB bus, or the USB port where you have it plugged into.

sdrtrunk is getting confused on if it can use the tuner or not.  

Recommend unplug and replug the tuner and manually adjust the frequency on the tuner a few times to make sure it's working correctly before you start the channel.

Also, you probably don't have to set the preferred tuner for the channel you're trying to run, when there's only a single tuner available.  If the tuner is working correctly, sdrtrunk will use it.

Varean, 2022-03-27:
I tried to go under 'Tuners' and select the RTL2832 SDR/R820T, but received this error, and the frequency settings are greyed out
[image]

I might be doing something wrong myself.

closed 2026-11-06 by DSheirer