DSheirer/sdrtrunk #1199: How many SDRs does SDRTrunk support

Brad-git-man, 2022-03-30:
I put 5 SDRs on the new version of SDRT (never tried it on older vers) and it made the data messages slow way down and have terrible reception. I have them on a 16 core PC and on a 3.1 USB port. Can anyone give any guidance on this?... I cannot find anywhere how many SDRs SDRT supports.

jamren01, 2022-03-30:
Please use the GoogleGroup for technical assistance. https://groups.google.com/forum/#!forum/sdrtrunk

Brad-git-man, 2022-03-30:
> Please use the GoogleGroup for technical assistance. https://groups.google.com/forum/#!forum/sdrtrunk

You know how many it supports, so why don't you have the courtesy to just say it????????????????????????????

liquidmayo, 2022-03-30:
I'm currently using 5 without issue. monitoring 2 p25 control channels and
one NBFM with squelch.   The only issue I have currently is the polyphase
channelizer error about how it can't change frequency fast enough or some
such, but rarely miss any traffic.

davidmckenzie, 2022-03-30:
Number of dongles isn't really a metric since it would vary so much depending on the systems being monitored and the computer running it. For example on my ancient macbook I can run 3 dongles with 1 system and 1 conventional net, but if I have the same number of dongles with 2 systems and 1 conventional I start missing frames and getting USB issues.

You just have to find the sweet spot for your particular setup

> You know how many it supports, so why don't you have the courtesy to just say it????????????????????????????

You might also want to consider being a little less rude to your fellow community members - this is an open source project after all.

sdr-stack, 2022-03-31:
With a decent computer: 5 with no issues, 6 works but is on the edge, like you can't adjust PPM once the channels are playing. There's work to split each tuner into its own thread which hopefully will bump the limit up. Linux and increasing the USB buffer size might go beyond 6 but I haven't tried, it does help with getting to ~5 but that system ran out of CPU to try any more.

echo 1024 > /sys/module/usbcore/parameters/usbfs_memory_mb

I'm guessing from my observations that a single USB controller might not be able to handle more than 6 and any other USB1/2 devices might lower the capacity of the controller you're using. Using USB3 hubs/ports won't increase this since they don't encapsulate USB2, it still runs over a single twisted pair shared among the entire system for that controller but Thunderbolt/Type-C docks might have their own USB controller which might encapsulate it.

Brad-git-man, 2022-03-31:
> With a decent computer: 5 with no issues, 6 works but is on the edge, like you can't adjust PPM once the channels are playing. There's work to split each tuner into its own thread which hopefully [...]

I have a 16 core PC with USB 3.1 ports & a USB3 hub w/ 5 tuners and in the Java "DOS Window" it says that "USB Bus Potential Maximum Data Rates {13MB/56MB} .... that seems like a disgraceful speed and I am wondering if there is any way to turn up the setting in the Java. I am the OP of this thread and I can't even three P25 channels and 2 VNF channels without dragging the message view speed down to a crawl and causing terrible reception on P25.

boteman, 2022-03-31:
Can't fix it if you don't know what's broke. We need metrics:
Is the CPU hammered?
Is the USB bus hammered?
Is a USB host controller not configured optimally?
Something else?

Brad-git-man, 2022-04-01:
> Can't fix it if you don't know what's broke. We need metrics: Is the CPU hammered? Is the USB bus hammered? Is a USB host controller not configured optimally? Something else?
. 
It is a Gigabyte X570 box that I built in 2019 and upgraded the CPU to a 16 core last yr. I have 5 SDRs in a powered USB 3 hub on a 3.1 port. ... I am now investigating the major USB speed issue on X570 that I forgot I read about in 2019.

DSheirer, 2022-04-17:
Once PR #1208 is merged to master you might check again the number of tuners that you can run at the same time.  Ths PR adds dedicated thread support for USB event processing to each tuner.

cfc62, 2022-04-30:
I've gotten up to 47 msps with 5 Airspy so far - 4 @ 10msps and 1 @ 7 msps and still running pretty well. 

8 cores @ 3.5 GHz and a dedicated PCi card for additional USB ports.

closed 2023-04-16 by DSheirer

ImagoTrigger, 2023-04-17:
Thanks Denny.  Common OS limits around 35MB/s via all/any USB channels/lanes  without "getting fancy"