DSheirer/sdrtrunk #1207: 0.5.0-b2/Win7x64: Should be possible to configure sound card as channel input source

powerbroker, 2022-04-16:
Following https://github.com/DSheirer/sdrtrunk/wiki/SoundCard_V0.3.0 within v0.5.0-b2 is impossible - there is no GUI to set sound card as channel input.

[image]

There should be a possibility to select sound card(e.g. default rec. device) to enable using of external scanners as radio source.

Brad-git-man, 2022-04-17:
Exactly, I had also noticed this and am wondering what happened to this feature? My area is switching to Phase II by 2024 and I would like to feed some you my Phase I GRE based scanners into PCs to decode Phase II, so they don't become antiquated junk.

DSheirer, 2022-04-24:
Sound card input to decoders is not supported in versions 0.4.0 or later.

closed 2022-04-24 by DSheirer