DSheirer/sdrtrunk #1225: MP3 stream corruption

amigan, 2022-05-01:
Using the latest HEAD cbeda19d27643910d0a15f8bb6f52e7425609325, I have noticed occasional "skipping" and garbling of the MP3 stream going through my icecast server. I also see errors when listening with mpv:
```
% mpv http://scanner/riscon
 (+) Audio --aid=1 (mp3 1ch 8000Hz)
AO: [pulse] 8000Hz mono 1ch float
A: 00:00:07 / 00:00:35 (21%) Cache: 27s/254KB
[ffmpeg/audio] mp3float: big_values too big
[ffmpeg/audio] mp3float: Error while decoding MPEG audio frame.
Error decoding audio.
A: 00:00:07 / 00:00:35 (21%) Cache: 27s/253KB
File tags:
 icy-title: Scanning...
A: 00:00:23 / 00:00:46 (50%) Cache: 22s/206KB
[ffmpeg/audio] mp3float: big_values too big
[ffmpeg/audio] mp3float: Error while decoding MPEG audio frame.
Error decoding audio.
A: 00:01:25 / 00:01:25 (100%) Cache: 0.0s
```

Device is RTL-SDR, system is P25 phase I. Built and running on a Raspberry Pi 4 running Manjaro/arm64 and liberica-jdk-full.

This was not occurring with SDRTrunk as of 18 February HEAD. I am happy to perform further troubleshooting or provide any requested information.

doug-hoffman, 2022-05-01:
I'm guessing by the URL that it's a self-hosted icecast server... What package/version is it?

Can you try going into the stream options and disabling "inline metadata"?

doug-hoffman, 2022-05-02:
I can confirm that inline metadata is broken with 2.4.0-kh15. It works fine with 2.3.2-kh32 (what Broadcastify uses) and 2.4.4, so it appears to be a bug in the 2.4.0-khX fork.

No issues with audio though, other than those caused by 2.4.0-kh failing to process inline metadata that result in metadata getting decoded as audio.

boteman, 2022-05-02:
The latest KH commit has a lot of "allow for setting a different default charset to utf8" messages, so maybe you must explicitly set the character set to avoid munging the metadata?

doug-hoffman, 2022-05-02:
Interesting find. I rebuilt kh15 with "ICY_CHARSET=latin1" and am waiting for some radio traffic to see if it made any difference.

doug-hoffman, 2022-05-02:
Nope...

```
$ mpv http://127.0.0.1:8000/stream
 (+) Audio --aid=1 (mp3 1ch 16000Hz)
AO: [coreaudio] 16000Hz mono 1ch floatp
A: 00:00:07 / 00:00:11 (65%) Cache: 3.9s/61KB
File tags:
 icy-title: 
A: 00:00:15 / 00:00:23 (69%) Cache: 6.9s/111KB
File tags:
 icy-title: Scanning...
A: 00:08:27 / 00:08:52 (95%) Cache: 24s/398KB
[ffmpeg/audio] mp3float: invalid block type
[ffmpeg/audio] mp3float: Error while decoding MPEG audio frame.
Error decoding audio.
A: 00:08:32 / 00:08:54 (96%) Cache: 22s/358KB
[ffmpeg/audio] mp3float: invalid new backstep -1
A: 00:08:44 / 00:09:03 (97%) Cache: 18s/298KB
[ffmpeg/audio] mp3float: big_values too big
[ffmpeg/audio] mp3float: Error while decoding MPEG audio frame.
Error decoding audio.
A: 00:08:49 / 00:09:07 (97%) Cache: 17s/280KB
```

doug-hoffman, 2022-05-02:
Found the likely cause...

icecast-2.4.4/src/format_mp3.c:
```c
    if (metadata)
    {
        state->inline_metadata_interval = atoi (metadata);
        if (state->inline_metadata_interval > 0)
        {
            state->offset = 0;
            plugin->get_buffer = mp3_get_filter_meta;
            state->interval = state->inline_metadata_interval;
        }
    }
```

icecast-2.4.0-kh15/src/format_mp3.c:
```c
    if (metadata)
    {
        client->flags |= CLIENT_META_INSTREAM;
        source_mp3->inline_metadata_interval = atoi (metadata);
        if (source_mp3->inline_metadata_interval > 0)
        {
            source_mp3->offset = 0;
            plugin->get_buffer = mp3_get_filter_meta;
            if (source_mp3->inline_metadata_interval > 7999 && source_mp3->inline_metadata_interval < 32001)
                source_mp3->interval = source_mp3->inline_metadata_interval;
            else
                source_mp3->interval = ICY_METADATA_INTERVAL;
            INFO2 ("icy metadata format expected on %s, interval %d", plugin->mount, source_mp3->interval);
        }
    }
```

The KH fork has decided to rewrite metadata intervals <=7999 to ICY_METADATA_INTERVAL (16000), changed in karlheyes/icecast-kh@fc3796ab2a470aa72ad764776afdbb7d1d25dd52. We have no way to know this and continue sending at the interval we specified.

We're using an interval of bitrate * 125 (2000 @ 16kbps, 4000 @ 32kbps). This likely means only 1 in 8/4 metadata updates gets treated as metadata, and the rest get passed as audio.

Bumping the interval to 8000 is not much of an issue for 128kbps streams, but 8000 bytes is a long time for frequently changing scanner audio at 16kbps.

Going to test a build with the interval set to 8000 to see if that makes any difference.

doug-hoffman, 2022-05-03:
Digging into the source a little further, ```source_mp3->inline_metadata_interval``` is the interval from the source and ```source_mp3->interval``` is the interval to downstream clients. It should still parse incoming metadata at the correct interval.

It's possible to override the outgoing metadata interval in the icecast configuration to work around the limit:
```xml
<mount type="normal">
    <mount-name>/stream</mount-name>
    <mp3-metadata-interval>2000</mp3-metadata-interval>
</mount>
```

The metadata works for a while, but eventually gets scrambled. The cause appears to be a transition from a long title to a short one:
```
[2022-05-02  20:44:19] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='Scanning...';...
[2022-05-02  20:44:21] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';...
[2022-05-02  20:44:21] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';...
[2022-05-02  20:44:22] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';...
[2022-05-02  20:44:23] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';...
[2022-05-02  20:44:44] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';ning...';...
[2022-05-02  20:44:51] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';UUStreamTitle='Scanning...';...
[2022-05-02  20:45:37] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';e='Scanning...';...
???X??05-02  20:48:18] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';!k??(?4
      O"@%StreamTitle='TO:200...
[2022-05-02  20:48:20] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';UUUUUStreamTitle='Scanning...';...
[2022-05-02  20:48:27] DBUG format-mp3/mp3_set_title icy metadata as StreamTitle='TO:11 Transport FROM:2307';UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU...
```

I have confirmed with packet captures that we are still sending valid metadata.

It will work seemingly forever if the metadata remains as 'Scanning...', but as soon as it changes to something else and back, things go wrong.

doug-hoffman, 2022-05-03:
2.4.0-kh10 appears to be the last version without the problem. Changes in kh11: https://github.com/karlheyes/icecast-kh/compare/icecast-2.4.0-kh10...icecast-2.4.0-kh11

doug-hoffman, 2022-12-17:
This issue is fixed by #1244. Currently, there is extraneous data being streamed from each MP3 file which results in icecast-kh getting out of sync with the metadata position. The new behavior only sends valid frames which doesn't trigger the bug in -kh.

DSheirer, 2022-12-18:
Closing - #1244 PR merged.

closed 2022-12-18 by DSheirer