DSheirer/sdrtrunk #1232: Exception in thread "AWT-EventQueue-0" org.usb4java.LibUsbException: USB error 9: error writing byte buffer: Pipe error

ryanthomp, 2022-05-04:
I get this error when trying to change anything about the tuner I currently have installed, I have tried reinstalling the drivers deleting the drivers moving USB ports however nothing seems to quite want to work. However, if I have 2 plugged in and neither start but I disable one of them a channel can sometimes start but once I stop it gives the same error. Any idea on how to fix this I know it isn't exactly SDRTrunks fault more or fewer USB drivers fault and my PC but help to troubleshoot this issue would be great.

Full Stack below
`Exception in thread "AWT-EventQueue-0" org.usb4java.LibUsbException: USB error 9: error writing byte buffer: Pipe error
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:798)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.writeDemodRegister(RTL2832TunerController.java:683)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.enableI2CRepeater(RTL2832TunerController.java:578)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController$ControllerAdapter.enableI2CRepeater(RTL2832TunerController.java:1504)
        at io.github.dsheirer.source.tuner.rtl.r820t.R820TEmbeddedTuner.setTunedFrequency(R820TEmbeddedTuner.java:144)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.setTunedFrequency(RTL2832TunerController.java:142)
        at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:191)
        at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:163)
        at io.github.dsheirer.source.tuner.TunerController.setFrequency(TunerController.java:228)
        at io.github.dsheirer.source.tuner.TunerController.process(TunerController.java:164)
        at io.github.dsheirer.gui.control.JFrequencyControl.fireSourceChanged(JFrequencyControl.java:187)
        at io.github.dsheirer.gui.control.JFrequencyControl$Digit.set(JFrequencyControl.java:380)
        at io.github.dsheirer.gui.control.JFrequencyControl$Digit.increment(JFrequencyControl.java:312)
        at io.github.dsheirer.gui.control.JFrequencyControl$Digit.increment(JFrequencyControl.java:305)
        at io.github.dsheirer.gui.control.JFrequencyControl$Digit$Listener.mouseClicked(JFrequencyControl.java:521)
        at java.desktop/java.awt.AWTEventMulticaster.mouseClicked(Unknown Source)
        at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
        at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
        at java.desktop/java.awt.Component.processEvent(Unknown Source)
        at java.desktop/java.awt.Container.processEvent(Unknown Source)
        at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
        at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
        at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
        at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
        at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
        at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)`

closed 2022-05-07 by ryanthomp

Brad-git-man, 2022-05-07:
Thanks for posting this issue. I am also getting this on one of my 5 Nooelecs, that I run all at once on SDRTrunk.

ryanthomp, 2022-05-07:
@Brad-git-man I solved this issue by downloading "UsbDriverTool" and going on my dongles and installing the realtek driver then going to device manager under sound devices uninstalling and disabling the device then once doing so I reinstalled WinUSB from Zadig if you need specifics I can do a full walkthrough if you need more help I was battling with this issue for quite a wile it's just a USB driver issue and windows being annoying.

Brad-git-man, 2022-05-07:
@ryanthomp what brand & how many SDRs are you running at once, just for comparative info to my 5.