DSheirer/sdrtrunk #1239: Tuner Editor View Spectrum Should Override Disabled Display

DSheirer, 2022-05-07:
Clicking View Spectrum in the tuner editor should override the disabled state.  Currently, right-click on the spectral display is the only way to override.

closed 2022-05-07 by DSheirer

GTR8000, 2022-05-07:
I'm seeing a bit of an issue with this.

If I disable the Spectrum & Waterfall in the main window, that setting now persists through restarts (thank you) as it did previously, so that's good.

However if I go to the Tuners tab, select a tuner, then click on New Spectrum Display, it's disabled unless I go back and click View Spectrum. The trouble is that it also re-enables the Spectrum & Waterfall in the main display, which I don't want.

Prior to the tuner redesign, you could leave the main window Spectrum & Waterfall display disabled permanently, while also being able to open up a separate window per-tuner. In other words, the two were independent of each other. Now they seem to be linked and it's a multi-step process to re-enable, open a separate window, then re-disable the main window display when you're done.

[image]

DSheirer, 2022-05-09:
@GTR8000 this should be resolved now with latest master.

PR #1240 (2022-05-07): #1239 Tuner Editor View Spectrum - Override Spectral Disabled State
Closes #1239
Resolves issue of view spectrum when the spectral display is disabled.  Tuner editor button overrides the property setting so that the view completes correctly.